atc1441/ATC_MiThermometer #1: Flexible ADV-timers.

vevsvevs, 2020-09-01:
First of all - thank you for your efforts, awesome work! Please, could you make ADV-timers more flexible to reduce battery drain? For instance, if temp/humi changed more than 1 degree/ 5 percent - send data every 1 min per change, and every 10 minutes otherwise (from the last send).

atc1441, 2020-09-01:
Thats a good idea, will look into it 👍

atc1441, 2020-09-01:
Just thought about it, it will not change much, as it always advertise the current data, and to get the new Sensor data it needs to ask the sensor for it which takes the most power as it uses clock stretching in the i2c interface.

It will update the LCD every 5 seconds and every 1minute will it put the new data in the Advertising buffer, but the buffer will be transmitted on every Advertising interval which is again at 1000ms. hard to explain, for my knowledge writing to the buffer does not draws more power so doing it every minute or 10 does not change much

TheNitek, 2020-09-01:
I think the idea was to advertise less often, but I would vote against it. In my use case I implemented some kind of timeout, so I rely on fix scheduled advertisments.

atc1441, 2020-09-01:
The stock firmware has an advertising interval of 1700ms is set it to 1000ms setting it to 1700ms would be no problem but the longer the distance is the harder it is to connect to the device if wanted. 

Increasing that value will indeed save power.

vevsvevs, 2020-09-01:
@atc1441 Hmm.. So, BLE-ADV sent out every second, and only data in it got actualized every minute? You think that BLE-stuff doesn't take too much energy being activated every second? I'm not quite sure (and I'll make test if necessary), but stock firmware advertise data only once per 10 minutes (5 if temp changed alot).

atc1441, 2020-09-01:
@vevsvevs as mentioned ble is active all the time otherwise you would not be able to connect to the device.

Also the stock firmware will advertise every 1700ms but then the data is just 08 and not the Sensor data

vevsvevs, 2020-09-01:
Ahh, I think I got it. Stock firmware should advertise every short time period to make this device discoverable. So BLE-stuff is activating anyway. So your implementation shouldn't drain battery much more than stock. Thanks for the info!

atc1441, 2020-09-01:
Will set the interval time higher now, just because 1000ms is still nearly double the power needed as 1700ms :)

i do a long run test with the custom firmware since this morning, since now the voltage didn't drop, it only increased as the battery was pushed while updating it.

vevsvevs, 2020-09-01:
Thank you! I'll try to make ESPHome integration for your data structure this night :)

atc1441, 2020-09-01:
Increased the Interval to now 1900ms and tested the update function, everything works as it should.

Update is on github :)

Will close here for now. please comment if any question are still there

closed 2020-09-01 by atc1441