atc1441/ATC_MiThermometer #2: Add Calibration/Offset

TheNitek, 2020-09-01:
One thing that bothered me for a while now is that it was not possible to calibrate the sensors, even though various sources (e.g https://github.com/JsBergbau/MiTemperature2#calibration ) mention that especially the humidity is not that accurate.
It would be awesome to be able to set an offset for temperature/humidity, which is then applied to the sensor data before displaying and/or broadcasting.

rufik, 2020-09-01:
SHTV3/SHTC3 sensor is pretty accurate according to datasheet, so there should be no need for calibration I think. RH measurement is not so simple/straightforward (as temperature for example), so consider it as approximate value like too dry - good - to moist :)

[image]
[image]

TheNitek, 2020-09-01:
In my experience (and also from what I have read, the lywsd03mmc has a tendency to show humidity 2%-5% above the actual value. Having a fixed offset would be nice to mitigate this (even if the error is not 100% linear, it should still be more accurate).

I would do it in my smart home implementation but I don't like the idea of having different value on the display and within my smart home.

atc1441, 2020-09-01:
It coulde be added and then writen to the flash. Need to experiment with flash writing a bit as it needs thinking about layout and position. Then it could set and read via a characteristic.

If its usefull or not is not so important i think as long as it helps the user experience.
for me the humidity is something i would not trust on its very position dependend as a simplefinger near by is already changeing the values.

What i did see is that the xiaomi value is a bit different to the custom one ca  bee seen in the video, the custom one against each other are very much the same values

TheNitek, 2020-09-01:
That might be a good observation. Maybe the Xiaomi firmware does some computing on the values, messing them up. I guess it makes sense to compare this a bit closer before implementing something.
I'll place two devices next to each other and compare values for a while.

TheNitek, 2020-09-02:
Ok so that's interesting. While the temperature stayed the same after flashing, the humidity dropped by 5% - So whatever the original firmware is doing, it doesn't seem does seem to alter the humidity quite a bit.

atc1441, 2020-09-04:
Added an offset for Temp and Humidity,

Temp from -12,5 - +12,5 °C

Humi -50 - +50%

The offset will be parsed as int_8 so an 0x80 would be -12,5°C something.

https://github.com/atc1441/ATC_MiThermometer#temp-and-humi-offset

closed 2020-09-04 by atc1441