atc1441/ATC_MiThermometer #4: Support other models like CGG1

Misiu, 2020-09-02:
First of all, wow, this looks amazing 🎉
I didn't know that custom firmware for this device was possible, thanks to you, now I know it is 🙂

I'd like to ask if it would be possible to support other devices, like CGG1?
There are multiple issues with it. Especially the advertising time.
It would be awesome if it could be set to one min and if the device could send all three parameters each time (now the battery isn't send too often).

A custom firmware could solve all the issues.

I'm not sure what MCU is used, but maybe similar.
Here are the materials I found: https://fccid.io/2AQ3F-CGG1/Internal-Photos/Internal-Photos-3990168
I can take my sensor apart and make some detailed pictures if needed.

atc1441, 2020-09-02:
Thank you

As far i know that one has an nRF51 something, its definitely also hackable but is a complete different route. Unfortunately it has the smallest flash size so its bit limited with the needed softdevice enabled but its doable as we can see on the device itself

I made some firmwares for nRF52832 Smartwatches so the knowledge is there 👍

Misiu, 2020-09-02:
@atc1441 awesome news 😁
If there will be an ESPHome integration for your custom firmware, then if someday a custom firmware for CGG1 will be made, then the same integration will be able to handle both models (in theory 🙂 )

atc1441, 2020-09-05:
Will close this for now as i get irritated by it. 

If there will be a custom firmware for other devices i will announce it :)

closed 2020-09-05 by atc1441