atc1441/ATC_MiThermometer #5: Battery performance

KoalaBear84, 2020-09-02:
Thanks for your work! You have some great knowledge / experience.

What I would like to know if the every minute part will drain the battery a lot more than the default implementation which does not send a lot of updates as I've seen. (looks maximum every 11 minutes if the value is different enough)

Yes, I'm aware the answer is still unclear! No problem, the issue can be left open until there is more info about it, and maybe added to the readme of this repo.

atc1441, 2020-09-02:
There is already an closed issue ablut that topic. 

The thing is that BLE is way more often active. For the Stock firmware every 1700ms and for the custom firmware every 1900ms in that period it will advertise its name and additional data like the advertising data.
On stock firmware the advertising data is only once every ten minutes the Sensor data and all the other time it is just 0x08 so even then it will be transmitted. Hope its understandable.

So in theory even changeing the advertising data every 5 seconds should not drain more power but i am not 100% shure about it

KoalaBear84, 2020-09-02:
Sorry about sort of a duplicate ticket. I indeed didn't check that out.

Wow, I didn't expect it was that active by default. It's insane that it could work for 6 months or more by stock firmware. With only deep sleep it could reach 4 years I check, according to your tweets 😂

I guess really sending a lot of updates would drain the battery more, but that is only a feeling, as I don't have any experience with all of this.

TheNitek, 2020-09-03:
Well there is one thing to consider: Sending more data in the advertisment means the transmitter has to send more data, which means more power consumption. So even if the adv frequency is the same, it might be bad for the battery if more data is sent.

atc1441, 2020-09-04:
That is a good catch, also here the stock firmware does advertise the service uuid all the time so it cant be that bad, they send about 8 bytes less when not sending the Sensor data so yes its not so much but still the rest.

What i could imagine what draws a lot is the measuring of the battery voltage. did set that to only do on advertising interval on even less, like every 5 minutes or so for now.

Just added it

atc1441, 2020-09-05:
Also added to display the battery level as percent on the LCD indicated via the battery symbol

closed 2020-09-05 by atc1441