atc1441/ATC_MiThermometer #6: Negative temperatures

leahneukirchen, 2020-09-02:
According to the README of https://github.com/JsBergbau/MiTemperature2 the sensor supports negative temperatures down to -9.9°C, but I think the code as-is doesn't work properly if that is the case.

I can offer to test with a freezer.

leahneukirchen, 2020-09-02:
Disregard, I misread that.

leahneukirchen, 2020-09-02:
Sorry, it says:

> When looking at the specifications this LYWSD03MMC Sensor is specified from 0 °C to 60 °C. The LYWSDCGQ (the Bluetooth Temperatur sensor with the round display and an AAA battery) is specified from -9.9. I can confirm this sensor also goes down to -9.9 °C. At colder temperatures it only shows an "L".

I think the *this* refers to the LYWSD03MMC indeed. So we should try that.

atc1441, 2020-09-02:
Oh you are right. Never tested it. Will look into it. 

Thank you

atc1441, 2020-09-03:
Added support for up to -9.9°C now :)

closed 2020-09-04 by atc1441

leahneukirchen, 2020-09-04:
Worked down to -16°C even, let's hope I can revive the display. :p

atc1441, 2020-09-04:
Yes in theory it can meassure lower than -9.9C but it simply can not display then anymore :) 

Its seems to me that your one has some fault as i didnt got any one that has this problem even after leaving it outside and having it in the freezer for a while

leahneukirchen, 2020-09-04:
Yeah, this one came back fine. :)

TheNitek, 2020-09-07:
README.md still says "Byte 11-12 Temperature in uint16", I guess that's not correct anymore?

atc1441, 2020-09-07:
fixing it, thanks for the tip

bangom, 2020-10-22:
> Yes in theory it can meassure lower than -9.9C but it simply can not display then anymore :)
> 
> Its seems to me that your one has some fault as i didnt got any one that has this problem even after leaving it outside and having it in the freezer for a while

Would it be possible to switch display to "precision of 2 and a scale of 0" below -9.9 degree? 

For example, -11.3 cloud be displayed as -11? BTLE data should advertise correct value -11.3.