atc1441/ATC_MiThermometer #7: After flashing firmware not able to see the device anymore

DoeWayne, 2020-09-02:
Great work you did, but after flashing the firmware I do not see the device anymore under Windows 10 device lists.
If I open your TelinkFlasher page the browser is able to find the device, but I'm not able to press DoActivation, I guess this could only be pressed once, right ? 

Do I have to made a note of the TOKEN and BIND KEY ?

Where do you get the original firmware ? Is it possible to get an original firmware read from the device as well over a webpage ?

Thanks

atc1441, 2020-09-02:
Hey  so the custom firmware does not need the Do Activation anymore. 

Also it does not need the token or bind key as everything is unencrypted.

You can if you want go back to stock with this firmware https://t.co/TsLExxEKJ3

DoeWayne, 2020-09-02:
Thx a lot, hwo did you get the original firmware ?

Do you know a good bluetooth application for windows, to trace traffic or to reveice the data.

atc1441, 2020-09-02:
The firmware can be captured while updating to get a download link.

But i was also able to dump it from the device with the genuine flasher. 

You need an application that support BLE

atc1441, 2020-09-02:
I have good experience with this one https://www.microsoft.com/de-de/p/bluetooth-le-explorer/9n0ztkf1qd98#activetab=pivot:overviewtab

DoeWayne, 2020-09-02:
Just for all other who are going back to the original firmware:

Status: Update done after 813.724 seconds

Now I can see the device as well under the bluetooth device list again. Thanks @atc1441 for all your support, we can close the request

closed 2020-09-02 by DoeWayne