atc1441/ATC_MiThermometer #8: Adjustable ADV-data buffer update period and smiley state parameter.

vevsvevs, 2020-09-03:
1 minute is cool for most common use, but maybe it's a good idea to implement adjustability for this parameter via Bluetooth, the same as it [was done](https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/ble.c#L77) for C/F setting? Also, it would be nice to have an adjustable parameter for smiley enabled/disabled/blink state ;)

atc1441, 2020-09-03:
The smiley blinking enable/disable is planned. 

But the interval would not help much to change. It does not save power if its longer or shorter. So i dont think its to usefull. Or why do you think its good?

vevsvevs, 2020-09-03:
Regarding smiley - I suggest implementing three states for it: permanently enabled, permanently disabled, and blink.

The interval changing goal is not a power consumption reduction (i got it from our previous conversation), but increase meaningful data obtaining frequency. In some circumstances, it's really handy to get actualized data every 10 secs.

atc1441, 2020-09-04:
A setting for the smiley an Advertising interval was now added 
https://github.com/atc1441/ATC_MiThermometer#settings-in-custom-firmware

closed 2020-09-04 by atc1441