atc1441/ATC_MiThermometer #9: Connection problems - again

DoeWayne, 2020-09-04:
Hello atc1411

just downloaded the new firmware and flashed the device, but now I get this:

Log:
17:59:01: Searching for devices
17:59:23: Connecting to: LYWSD03MMC
17:59:33: Detected Mi Thermometer
17:59:33: Connected
17:59:44: Disconnected.
17:59:44: Searching for devices
18:00:08: Connecting to: LYWSD03MMC
18:00:38: Detected Mi Thermometer
18:00:38: Connected
18:02:07: Activating now, please wait...
18:02:10: Activation successfull
18:02:11: Received device infos are correct
18:02:11: Login successfull
18:03:03: File was selected, size: 61452 bytes
18:03:03: Count: 3841
18:03:11: Start DFU
18:04:07: Update done after 55.577 seconds
18:04:14: Disconnected.
18:04:21: Searching for devices
18:04:43: Connecting to: LYWSD03MMC
18:05:06: Detected Mi Thermometer
18:05:06: NotSupportedError: GATT operation failed for unknown reason.
18:05:06: Reconnect 1 from 5
18:05:06: Detected Mi Thermometer
18:05:06: NotSupportedError: GATT operation failed for unknown reason.
18:05:06: Reconnect 2 from 5
18:05:06: Detected Mi Thermometer
18:05:06: NotSupportedError: GATT operation failed for unknown reason.
18:05:06: Reconnect 3 from 5
18:05:06: Detected Mi Thermometer
18:05:06: NotSupportedError: GATT operation failed for unknown reason.
18:05:06: Reconnect 4 from 5
18:05:06: Detected Mi Thermometer
18:05:06: NotSupportedError: GATT operation failed for unknown reason.
18:05:06: Reconnect 5 from 5
18:05:06: Detected Mi Thermometer
18:05:06: NotSupportedError: GATT operation failed for unknown reason.
18:05:06: Something went wrong, to many reconnect's

Great that you implemented the new features to calibrate, but how can these be set, what kind of application is required, or di I misunderstood something here and it need to be set in the firmware (which I do not believe):

Settings in custom firmware:
The following settings can be send to the RxTx Characteristics 0x1F10/0x1f1f These settings will not get saved on power loss, maybe that will change in future but normaly the battery will be in there for a while

Change display to °F or °C:
0xFF = Lcd in °F

0xCC = Lcd in °C <- Default

Blinking smiley:
0xA0 = Smiley off

0xA1 = Smiley happy

0xA2 = Smiley sad

0xAB = Smiley bliking <- Default

Advertising interval
byte0 0xFE

byte1 0x06 - value times 10 seconds = interval 60 seconds default.

Temp and Humi offset
byte0 0xFA = Temp offset

byte0 0xFB = Humi offset

byte1 as an int8_t

so Temp = range -12,5 - + 12,5 °C offset Humi = range -50 - +50 % offset

The offset will be parsed as int_8 so an 0x80 would be -12,5°C something.

atc1441, 2020-09-04:
Please try to reboot the device and reload the web flasher and try again to connect. 

You can use nRFconnect to do the settings. 

Also now that you say it i should add a function to set the settings via the web flasher.

DoeWayne, 2020-09-04:
Also now that you say it i should add a function to set the settings via the web flasher. 

That would be great.

I've rebooted the device, but still getting the same error.

atc1441, 2020-09-04:
Please try it on another device if possible

DoeWayne, 2020-09-04:
I've tried it with another device which do not have the custom firmware installed, there it works. So now I've tried the other one again and I get the same messages, but if I press reconnect I got:

20:30:03: Reconnect
20:30:07: Disconnected.

But the status on top still mentioned:

Status: waiting for you to connect a device
Temp/Humi: waiting for data

atc1441, 2020-09-04:
There should not happen more on the custom firmware right now as its just meant to flash the device. Please try to flash it again when its connected

RadicalGR1, 2020-09-04:
Hallo I want to connect that sensor on a raspberry pi 4 with Hass.io the problem is that the Hassio identifies the bt divices only when it starts.... To identify my smartphone p.a. I have to scan for bt divices on my phone continuously as long as the Hassio boot! How can I force the sensor to transmit so I can identify it in Hassio? Sorry for my bad English?

DoeWayne, 2020-09-04:
Not possible 

Log:
21:06:47: Searching for devices
21:07:15: NotFoundError: User cancelled the requestDevice() chooser.
21:07:15: Reconnect 1 from 5
21:07:20: Searching for devices
21:07:33: Connecting to: ATC_*****C
21:08:26: Detected Mi Thermometer
21:08:27: NotSupportedError: GATT operation failed for unknown reason.
21:08:27: Reconnect 1 from 5
21:08:27: Detected Mi Thermometer
21:08:28: NotSupportedError: GATT operation failed for unknown reason.
21:08:28: Reconnect 2 from 5
21:08:28: Detected Mi Thermometer
21:08:28: NotSupportedError: GATT operation failed for unknown reason.
21:08:28: Reconnect 3 from 5
21:08:28: Detected Mi Thermometer
21:08:28: NotSupportedError: GATT operation failed for unknown reason.
21:08:28: Reconnect 4 from 5
21:08:28: Detected Mi Thermometer
21:08:29: NotSupportedError: GATT operation failed for unknown reason.
21:08:29: Reconnect 5 from 5
21:08:29: Detected Mi Thermometer
21:08:30: NotSupportedError: GATT operation failed for unknown reason.
21:08:30: Something went wrong, to many reconnect's
21:08:41: File was selected, size: 61452 bytes
21:08:41: Count: 3841
21:08:46: Start DFU
21:08:46: Update error: TypeError: Cannot read property 'writeValue' of null

atc1441, 2020-09-04:
Something is off there.
It should not detect an Mi Thermometer. 
But i got it now. It was the latest change will write when i solved it

atc1441, 2020-09-04:
Please retry it now and report :)

DoeWayne, 2020-09-04:
no same problem.

atc1441, 2020-09-04:
Mhh that is very strange, i tried it with multiple devices, and even with the bug i remove now it worked for me, it sounds like something has gone wrong while flashing the new firmware.

did you tried to flash another device?  is the Mi still being detected on the custom firmware? what Os do you use, what browser? 

But in addition i added some settings to the flasher to set the custom firmware

DoeWayne, 2020-09-04:
Do not aks me what has been changed, but I tried it now multiple time, also removed the batterie for a long time nothing changed. I just tried it again, and now the toll mentioned:

Log:
00:28:57: Searching for devices
00:29:07: Connecting to: ATC_*****C
00:29:14: Detected custom Firmware

I'm now able to sent the new setting, expect the offsets, they seems not to work.

DoeWayne, 2020-09-04:
Forgot to mention, I entered 

FA+0x80

to increase the temperature by 12.8 ! Is that correct ? If i want to decrease the value I have to enter a negative value, right ?

atc1441, 2020-09-05:
The first bit indicates if its negative or positive.

So 0x80 0b10000000 should be -12,5 about

0x7f 0b01111111 should be +12,5 about

You can use this one for example to convert it.
https://cryptii.com/pipes/integer-encoder

But i will change set site later to have that simpler

DoeWayne, 2020-09-05:
I love your work and support, this is absolute awesome, thx

I used that site already, but thx for the link this might be usefull for all others

atc1441, 2020-09-05:
It is how an int8 is represented in binary at so 0b10000000 = 0x80 = -128 so -12,8°C offset

DoeWayne, 2020-09-05:
ok thats for minus, but how to get the plus calibration for 12.8°C or 35.8°C

atc1441, 2020-09-05:
that would be the equivalent 0b01111111 = 0x7F = 128 = +12,8°C offset

please look at some basic infos on how integers and unsigned integers work that will help you understand it more.

DoeWayne, 2020-09-05:
thanks

What do you mean with Advertising Type: Custom and Mi Like

atc1441, 2020-09-05:
Please take a look at this issue https://github.com/atc1441/ATC_MiThermometer/issues/3

closed 2020-09-05 by atc1441

Sharafanovich, 2023-08-21:
у  меня тоже пишет эту ошибку на 2х устройствах, что делать?
**Update error: TypeError: Cannot read property 'writeValue' of null**