atc1441/ATC_MiThermometer #11: LCD going off after some time

6Xkljk8Vj7Tk1k4, 2020-09-05:
Hi,

After some time (like one-two minutes) LCD is going off - however it is still possible to connect/send firmware using TelinkFlasher.
Is it a bug or feature?

Best Regards

atc1441, 2020-09-05:
Hey, it seems @leahneukirchen also hat that problem with one Thermometer,

did you try to reinsert the battery? 

there is one unknown pin goin to the lcd controller and it may have something to do with that.

my devices all work fine so its hard for me to debug

leahneukirchen, 2020-09-05:
I have one running your firmware, which went out in the freezer and came back fine later.

I have one running on stock firmware again which shows this behavior, it went out after 5-30s with your firmware after power cycling.

In general, power cycling made the display work again, but then it went off with the latter one again.

atc1441, 2020-09-05:
Do any one of you may have the possibility to open one ? 

that way its could be compared against the other ones to see if they are a different revision

leahneukirchen, 2020-09-05:
I need to find such a small torx, maybe I have one at work. Gonna take a few days.

I ordered both in the same shipment, I can compare the numbers on the back next wekk.

6Xkljk8Vj7Tk1k4, 2020-09-05:
I took few more from my order and they work as expected. I had a problem with one more, however after turn off and on again it shown temperature. Anyway the first one is still off, switching on/off, reflash or battery replace didnt help. 
[board1](https://ibb.co/7Vpm7Hp)
[board2](https://ibb.co/4WfdY12)

[:)](https://ibb.co/yWgCWrq)

atc1441, 2020-09-05:
Thank you for the images. My boards are exactly the same.

Must be something else

benek984, 2020-09-05:
I have flashed my 14 sensors and all are experience that same issue. After several minutes screen freeze and doesn't display current temperature correctly. Power cycle fixing the issue but just for a another several minutes. I haven't have this problem on the stock firmware.

bbbenji, 2020-09-06:
I flashed 4 of my sensors, and one of them is experiencing the aforementioned screen blanking. Reinserting the battery fixes it, but only temporarily. It may have to do with the fact I use that one in the fridge as I pulled it out this morning and it hasn't blanked yet. I will swap it with another and see if I can replicate the issue.

benek984, 2020-09-06:
I have flashed my 14 sensors again this morning using the latest .bin file. After flashing I set them to as follow:
Smiley: OFF
Advertising Type: Mi Like
Show battery in LCD: Disabled
Advertising interval: 10 Seconds 

So far all good, sensors doesn't freeze anymore and reporting back to my Home Assistant by connecting via ESPHome BLE_tracker and xiaomi_cgg1 platform.

example of my configuration:

esp32_ble_tracker:
sensor:  
  - platform: xiaomi_cgg1
    mac_address: "A4:C1:38:BD:89:4E"
    temperature:
      name: "Bedroom Temperature"
    humidity:
      name: "Bedroom Humidity"
    battery_level:
      name: "Bedroom Battery Level"

bbbenji, 2020-09-06:
Reporting back that the thermometer that was working fine for days now, blanked after 30min of being in the fridge. I suspect it has to do with low temperatures. Will try another thermometer.

benek984, 2020-09-06:
> Reporting back that the thermometer that was working fine for days now, blanked after 30min of being in the fridge. I suspect it has to do with low temperatures. Will try another thermometer.

try my settings:

Smiley: OFF
Show battery in LCD: Disabled

This prevent sensor on blinking between two screens (battery or humidity) in my case those settings have solved the issue.

bbbenji, 2020-09-07:
@benek984 Unfortunately those settings did not help. I can quite consistency reproduce the screen blanking by placing any of my 4 thermometers in the fridge. Need to try using ATC format, using Mi at the moment.

atc1441, 2020-09-07:
@bbbenji ok the not showing of data on the lcd when in freezer is also same for me. Havent tested that with stock firmware but i think that is display related. If i press on the lcd when in freezer it comes back while pressing it. 

Dies this work as well?

bbbenji, 2020-09-07:
Pressing the screen does not do anything for me. Sounds like a hardware issue. However, my fridge is approx. +6 degrees  celsius, well within the spec'ed range. I have not tested stock firmware yet.

atc1441, 2020-09-08:
Tested a bit further, Stock vs. Custom rom in freezer, the display of the Stock rom stays on the custom not
[image]

Hard to do an i2c capture while in freezer, i could imagine they set the voltage of the lcd controller higher when under a specific °C

atc1441, 2020-09-08:
Looking further onto the PCB i following the unknown pin i already mentioned and will try to capture if that one has something to do with it.

[image]

atc1441, 2020-09-08:
[image]

Good news, setting that pin (PB6) to a level and not leaving it floating fixed the issue :)
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/main.c#L30

Its already updated in the repo and can be flashed.

nikito7, 2020-09-09:
Still an issue at 26° on one sensor

atc1441, 2020-09-09:
Thank you for the info. 
Is it instant off after inserting the battery or how is the behavior?

nikito7, 2020-09-09:
After new flash. When I connect
I waiting 60s its ok, then connect with flasher turn off.

Maybe related with some ble discovery

Because now, after insert battery, goes off 10-20s after

nikito7, 2020-09-09:
Now after blink one time
28°
Maybe temperature related.

I flashed stock again, and then custom, I never reached the 60s in first time

nikito7, 2020-09-09:
At 17°C works fine....

leahneukirchen, 2020-09-09:
I flashed the firmware of HEAD on the Mi that went dark quickly, and it turns off immediately after the MAC now... Bluetooth works fine.

I also opened it and wiggled the LCD connector a bit, but I think that is not the issue, as it displays fine in the begining.

(Display was fine with the stock firmware, too.)

nikito7, 2020-09-09:
Maybe are clones
I have one with a "very soft purple" background lcd

leahneukirchen, 2020-09-09:
Indeed, the LCD colors are different. The working one is "very soft purple", the non-working one is "dark greenish".

nikito7, 2020-09-09:
The problem is above 20°C

It's working at -10°C (-9.9 on lcd)

Cold kills battery fast :)

leahneukirchen, 2020-09-09:
It was only cold for a few minutes. But we have 20°C inside, so....

nikito7, 2020-09-09:
Maybe less

Put in freezer, remove/reinsert the battery less 15° because hot hands

My works that way. If I take it out freezer. Go blank after a while

leahneukirchen, 2020-09-09:
Powercycled it at 11.8°C, display came back; warmed up in ~15min to 20°C. Still on.. let's see.

leahneukirchen, 2020-09-09:
Went out at ~20.5°C again. :(

nikito7, 2020-09-09:
Maybe is the Humidity...

21.3°C 78% still on

leahneukirchen, 2020-09-09:
Humidity fell to 60%.

atc1441, 2020-09-09:
I could also imagine that the display needs to be set into a different mode on different temperatures. So like pull pin low on low temp and high on high temp. 

A reinit of the lcd is needed after setting a different level on that pin. But that shouldnt be a problem to do.

To bad there is no info on the lcd controller

atc1441, 2020-09-09:
On mine i had only problems on very low temp so far maybe someone can upload an comparison of the two lcds in one picture

leahneukirchen, 2020-09-09:
Perhaps expose the pin over bluetooth so we noobs can toggle it over the browser. :)

nikito7, 2020-09-09:
10s broadcast maybe faulty

the event:
[image]

atc1441, 2020-09-09:
@leahneukirchen only toggle would help to much as the init is needed. So needed compine that somehow first :-/

@nikito7 can you explain that a bit more? 
I can imagine that the newly added alarm do somethong wrong there, it will do an instant advertise if the Temp or Humi changes more than 0.5C / 5% between a main loop delay (5-6 seconds), there is a bug in it that will advertise twice if in the same loop a timed advertising was happening. 
Will change that

atc1441, 2020-09-09:
@nikito7  fixed that particular bug right now

nikito7, 2020-09-09:
chart was for lcd off event

nikito7, 2020-09-09:
Your change fixed LCD issue

atc1441, 2020-09-09:
Yeah should not fixed that one. Only changed the instant advertising. 

Will look deeper into what the stock firmware is doing with the pin. Maybe even capture some data while im freezer with a wire to a logic analyzer or something.

nikito7, 2020-09-09:
Can you confirm that 10s broadcast is working?

If working. Esphome or homeassistant is filtering data

atc1441, 2020-09-09:
@nikito7 yes can confirm that it does work at 10s interval just tested it again. Its 10-15 Seconds to be precise as its only tested for the interval if the main loop interval was overflowed as well.

There is a problem with interval bigger than 3minutes but that will be fixed later.

Calcu2006, 2020-09-10:
Hi benek984

I did everything like you, but maybe i'm forgetting something... i can read the values in MQTT.. can you please put an example about how are you reading it?

I created this sensor:
  - platform: mqtt
    name: "testing pc"
    state_topic: "esp_entrada/sensor/temperatura_ordenador/state"
    unit_of_measurement: "°C"

but status is always unknown..., what i'm missing ?

Thanks for answering!

> I have flashed my 14 sensors again this morning using the latest .bin file. After flashing I set them to as follow:
> Smiley: OFF
> Advertising Type: Mi Like
> Show battery in LCD: Disabled
> Advertising [...]

nikito7, 2020-09-10:
> Hi benek984
>
> I did everything like you, but maybe i'm forgetting something... i can read the values in MQTT.. can you please put an example about how are you reading it?
>
> I created this sensor:
>
> * [...]

Use api in esphome.yaml

api:
    password: pass

And add integration "esphome" in homeassistant

benek984, 2020-09-11:
Hi Calcu2006,

I use ESPHome plugin with Home Assistant, to read values from sensors I have ESP32 board with ESPHome on board. 
As mentioned by nikito7 this will autodiscover sensors added to the ESP32 config so you don't need to edit your yaml config file in home assistant. 
Example config below:

esphome:
  name: esp32_ble_01
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xxxx"
  password: "xxx"
  power_save_mode: none
  use_address: 192.168.10.135
  

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxx"
    password: "xxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API

api:
  password: "xxx"

ota:
  password: "xxx"
  
web_server:

# Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:

sensor:
      
  - platform: xiaomi_lywsd03mmc
    bindkey: "26d11deb1fcfae7862a462839a5156a9"  ----- "This is fake value but is necessary for that platform, you can use diffrent xiaomi platform from ESPHome but I found this one the most stable for those sensors". 

    mac_address: "aa:aa:aa:aa:aa:aa"
    temperature:
      name: "Bedroom Temperature"
    humidity:
      name: "Bedroom Humidity"
    battery_level:
      name: "Bedroom Batery Level"

Calcu2006, 2020-09-11:
Thanks, i was creating MQtt sensors... now everything works, thanks again !

rnizametdinov, 2020-09-15:
Same issue - revert to stock firmware

giannoug, 2020-09-16:
Same issue here. Right after flashing, the boot screen appears (showing "Atc", happy face and mac), then screen turns off. This is what happens all of the time, but it stayed a couple of seconds in the temperature / humidity screen once.

atc1441, 2020-09-16:
I compiled a version now where it pulls the LCD pin high and not low, maybe someone who has the error can flash it and test it, that way we would know if looking into that direction would help.
especially @rnizametdinov  and @giannoug 

[ATC_Thermometer_LCD_HIGH.zip](https://github.com/atc1441/ATC_MiThermometer/files/5235115/ATC_Thermometer_LCD_HIGH.zip)

giannoug, 2020-09-16:
@atc1441 Just flashed `ATC_Thermometer_LCD_HIGH.bin`. Doesn't seem to fix the issue. It goes to blank screen straight after booting

atc1441, 2020-09-16:
Thank you for testing it. Looks like its still a different error.

Are the other ones from you working fine?

giannoug, 2020-09-16:
@atc1441 What other ones are you referring to? I don't have other devices available and the stock firmware works fine (both the one that came from the manufacturer and the one you provide in the README)

atc1441, 2020-09-16:
Thought you maybe bought a pack of four or something. 

But no problem :)

nikito7, 2020-09-16:
The good ones have lcd issue now
Dont update :p

atc1441, 2020-09-17:
@nikito7 yes that was just for the bad ones as a test :)

Will delete the file now.

rnizametdinov, 2020-09-17:
> especially @rnizametdinov and @giannoug
> Edit. File deleted to prevent miss flashes
I have 3 thermometer.
2 of them - turn off screen on this firmware. 
And last one - work 3 hours and everything OK. 
I tray change battery between them - everything same: first 2 screen off, last one - work fine.
I will tray open them and find differense.

rnizametdinov, 2020-09-17:
> Thank you

[image]
[image]

Left - turn off screen. As I can see - they are the same :(. 
I try use good motherboard with "bad" display - work well. 
I try use bad motherboard with "good" display -  don'twork. 
On motherboard I see two connection roads from main controller to display controller - is it use i2c for display control?

nikito7, 2020-09-17:
Why chip numbers change?

[image]
[image]
[image]
[image]

atc1441, 2020-09-17:
Thank you for the in depth pictures, it seems like they changed something in the LCD controller ic then, 

All my have the F200 version. and they work fine

[36 more replies]

closed 2020-11-12 by atc1441