atc1441/ATC_MiThermometer #12: Immediately ADV-data updates in case of too fast values changing.

vevsvevs, 2020-09-06:
Thanks to your efforts, this device became absolutely perfect, imo! One more suggestion: please implement the immediate message sending in case of temp/humi change hit some threshold in a certain period of time. I mean, something like this: if temp changed for 0.5C (adjustable by user), or humi changed for 5% (adjustable too) since the last sensor read - ADV-message buffer updates immediately, regardless of the timer state. I know that it could be done easily with ESPHome in conjunction with a pretty short set timer, but maybe other smart home system integrations can't do this, so internal hardware implementation would be useful.

atc1441, 2020-09-07:
Thought about a simple way of implementing this but havent come up with a good solution so far.

Will most likely take some time till its implemented, had begun now with having the old values available, https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app.c#L85

It should simply compare the last with current and if its more or less then the set threshold it does an advertising so a simple if statement

i will implement it so it will make an advertising out of the regular interval but still have the normal interval not being interrupted by it so the interval is still on the same time every minute/ like its set.

vevsvevs, 2020-09-07:
Yep, that's exactly what I meant above (plus make intervals adjustable) :) Thanks a lot for your work!

atc1441, 2020-09-08:
Good.

The interval is already set able 👍 even if there is a bug so you only can set it to max 3 minute round about but will be fixed one day

atc1441, 2020-09-08:
Added the feature now, its Default to 0,5°C and 5% 

https://github.com/atc1441/ATC_MiThermometer#temp-or-humi-instant-advertising

closed 2020-09-08 by atc1441