atc1441/ATC_MiThermometer #18: possible wrong battery reported

xuefer, 2020-09-12:
HA says 100% using official firmware, while custom firmware shows minimal battery icon on LCD

atc1441, 2020-09-12:
Hey, 

so the stock battery reporting is broken. 

The reason you see the Battery icon in the LCD is because i use it to show the current battery % in the Humidity display

Every time the Battery icon is showing the current status is viewable in the Humidity field. you can turn that of via the Web Flasher if not needed.

Also i set the 100% battery to 3.1V and not to 3.0V so you can better see how much overhang the battery still has.

xuefer, 2020-09-12:
thanks. it's a good feature except for lacking of document. i'll keep it on despite it's blinking

closed 2020-09-12 by xuefer

atc1441, 2020-09-12:
Good :) 

It is documented here: https://github.com/atc1441/ATC_MiThermometer#show-battery-level-in-lcd-

xuefer, 2020-09-12:
indeed, but hard to understand
```
The battery level will be shown on the LCD every 5-6secdonds indicated by the battery symbol at the humidity display.
```

```
Will show Humidity% or Battery% flipping back and forth every 5~6 seconds, with battery symbol at bottom indicated it's battery level
```
i'm not English native though, but i do believe the meaning of "switching"/"flipping" should be emphasized

atc1441, 2020-09-12:
Simply added both versions for now.

Not an English native myself so i dont know better :)