atc1441/ATC_MiThermometer #19: Advertising interval not honored

fitch, 2020-09-15:
At first, thanks for amazing work on this flasher & custom firmware! :)

I have 6 devices flashed with the newest https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer.bin. It seems that despite specifying the advertising interval to 1 minutes, I seem to be getting advertisements quite often (every 1-20 seconds). Is there something I'm missing?

atc1441, 2020-09-15:
Hi, 

currently the advertising is limited to 3 minutes maximum but should work for 1 minute fine. 

also it does notify when a change greater than 0.5°C or 5% data is happening then it does notify instantly, you can change that behavior with the web flasher.

soloam, 2020-09-16:
Hello @atc1441  so setting to 10s does not work correct? I also tried to set FC01 to notify at 0.1ºC change, but does not seems to work!

Tks

atc1441, 2020-09-16:
Hey. Both thinks should work fine

soloam, 2020-09-16:
I'm using esphome to grab the changes. I do it with "Mi Like", the 10s passes and no update, also I blow hot air to the sensor, I see the value changing in the lcd, but no report.

Thank You

atc1441, 2020-09-16:
It is possible that esphome has it blocking for faster updates?!

nikito7, 2020-09-16:
mmc only advertises every 10min but multiple times shortly

try with platform xiaomi_cgg1

kktmp1, 2020-09-30:
> Hi,
> 
> currently the advertising is limited to 3 minutes maximum but should work for 1 minute fine.
> 
> also it does notify when a change greater than 0.5°C or 5% data is happening then it does notify instantly, you can change that behavior with the web flasher.

@atc1441 It is possible to increase the Advertising interval above 3 minutes. The battery is draining fast. Using the FEFF command wouldn't be useful, would it?

Thanks for your amazing work !!

atc1441, 2020-09-30:
The battery should not drain faster, it is doing the same thing even with short interval. 

in general the battery is shown lower as it is as 100% is 3.1V and not 3.0V in my firmware to see how much overhand there is

kktmp1, 2020-09-30:
Ok thanks @atc1441  I'll keep an eye on it

Of 8 sensors, an average value of 80% in two weeks. It may be the expected normal battery discharge.

closed 2020-10-02 by atc1441