atc1441/ATC_MiThermometer #21: Is this supposed to work from IOS?

G-Christ, 2020-09-17:
Hi
Do you know whether the flasher works from iOS? 
when I push the connect button  on mig iphone, it writes "searching for devices” in the log and nothing more happens.

atc1441, 2020-09-17:
So far ios is not capable of WebBluetooth sadly

ghoeffner, 2020-09-17:
If you have a recent notebook with chrome at hand, it should work there. Works with a T570, Win 10 and Chrome.

G-Christ, 2020-09-17:
Thanks it works 😊
Is it correct it can not just show the bind_key generated by and works wirh the Mi home App, but it generates a new one, and then the sensor will no longer work with the  mi home app?

Med venlig hilsen
Ghassan Christensen

atc1441, 2020-09-18:
That is correct. 

After activating it in the web flasher you need to re register it to Mi Home which makes the Web flasher key worthless again.
Only one works at a time

G-Christ, 2020-09-18:
Ok thanks,

What about integration to Hassio?
Is it possible to integrate the flashed sensor in some way in to Hassio?
Maybe by choosing advertising type “Mi like” , or will the device type be changed too?

If not do you know if it is possible to modify original firmware  so that sensor data are posted/advertised more often than the current every 10 min?

Med venlig hilsen
Ghassan Christensen

atc1441, 2020-09-18:
I dont know Hassio but with Mi Like it is most likely.

Its not simple to change the stock firmware to do that.

G-Christ, 2020-09-18:
Ok, I think I will give a try with Mi like

I’ll let you know whether it works or not

Thanks for your great work

Med venlig hilsen
Ghassan Christensen

closed 2020-09-20 by atc1441

atc1441, 2020-10-10:
That sounds like the laptop does not have BLE or it is disabled.

Ypu can try this app to see if it has BLE https://www.microsoft.com/de-de/p/bluetooth-le-explorer/9n0ztkf1qd98?activetab=pivot:overviewtab

zuidwijk, 2020-10-10:
> That sounds like the laptop does not have BLE or it is disabled.
> 
> Ypu can try this app to see if it has BLE https://www.microsoft.com/de-de/p/bluetooth-le-explorer/9n0ztkf1qd98?activetab=pivot:overviewtab

Well I don't have Windows, I do have Mac OS (which I think doesn't work) and I have Ubuntu. And under Ubuntu I have bluetooth enabled, as I can connect my keyboard, mouse and network dongle (all 3 bluetooth).

atc1441, 2020-10-10:
Ok then this is interesting for you 

https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md

zuidwijk, 2020-10-10:
> Ok then this is interesting for you 
> 
> 
> 
> https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md

Hero! Even works on my mac 👍🏼