atc1441/ATC_MiThermometer #22: Can't reconnect to flashed thermother

ppmt, 2020-09-20:
I have managed to flash one of my sensor with the custom firmware all worked
But now I can't reconnect to it.
I can't see it but when I select it it end up saying in the log:

17:47:17: Searching for devices
17:47:29: TypeError: device.watchAdvertisements is not a function
17:47:29: Reconnect 1 from 5
17:47:34: NetworkError: Connection failed for unknown reason.
17:47:34: Reconnect 2 from 5
17:47:40: NetworkError: Connection failed for unknown reason.
17:47:40: Reconnect 3 from 5
17:47:46: NetworkError: Connection failed for unknown reason.
17:47:46: Reconnect 4 from 5
17:47:51: NetworkError: Connection failed for unknown reason.
17:47:51: Reconnect 5 from 5
17:47:56: NetworkError: Connection failed for unknown reason.
17:47:56: Something went wrong, to many reconnect's

I have no issue with another sensor that has not been flash yet. I don't dare flashing it :)

atc1441, 2020-09-20:
Please try restarting pc and also take out battery from the thermometer for a bit.
Maybe also try it with another device

ppmt, 2020-09-20:
When you say another device you mean another PC? For some reason the only way I can get you program to work is on a chromebook. Any other device I tried (android phone, linux laptop) won't work. The windows showing you the available bluetooth  devices.
I have another thermometer still on original firmware and this one does show up with no issue on the chromebook. I don't really want to reflash until the first one can connect again. 

Will try to reboot the chromebook and remove the battery

atc1441, 2020-09-20:
Yes sorry meant another pc or mobile device. but looks like you already did that.

It can also happen that the flashing went wrong. is so it can be recovered by the USB to UART converter method but would be better of course if its possible to revive it without that.

ppmt, 2020-09-20:
Ok some progress. First thank for your help.
I restarted the chromebook and removed the battery. I can now see that in the logs it says:
Detected custom Firmware.

Now where it says Temp/Hum I can see waiting for data
It has been several minutes now and it won't populate.

Regarding the flashing I would guess it went well because the thermometer itself is working fine and displaying the temp/humi/battery
I have  now been able to reflash it several time. So I think we can close this case. I now need to rewatch your video again to understand how to get the bind key so I can integrate it in ESPhome

Thanks a lot!

ralf-e, 2020-09-20:
> Any other device I tried (android phone, linux laptop) won't work. 

Which browser are you using. With Google Chrome, it should also work on Android or Windows

atc1441, 2020-09-20:
Yes that definitely spunds like it worked.

On custom rom the temp and humi will not be shown 

Will close it now 👍

closed 2020-09-20 by atc1441

ppmt, 2020-09-20:
> > Any other device I tried (android phone, linux laptop) won't work.
> 
> Which browser are you using. With Google Chrome, it should also work on Android or Windows

I am on Linux using Firefox. I tried with Google Chrome on Linux and it would not work either. Don't really want google chrome on my phone so will stick to the chromebook for now :)