atc1441/ATC_MiThermometer #24: MHO-C401 just web log

nikito7, 2020-09-24:
...
Log:
04:37:00: Searching for devices
04:37:03: TypeError: device.watchAdvertisements is not a function
04:37:03: Reconnect 1 from 5
04:37:05: Detected Mi Thermometer
04:37:05: Connected
04:37:21: Activating now, please wait...
04:37:25: Activation successfull
04:37:26: Received device infos are correct
04:37:26: Login successfull
...

Device known id:
blt.3.129v87p7vBATC

Mi Token:
6f6faa84c6b69c2d9c1fd0c7

Mi Bind Key:
fc683bbbc65eceeb9d0ce9dd9ec68ebd

Dont work in esphome, neither MiHome now xD

"Verification failed"

atc1441, 2020-09-24:
It seems like the device was not activated before as it was named a custom ID.

It is very likely that the MiHome app gives another device ID and checks later if it fits.

To delete the device id it would be needed now to either flash the stock firmware via the USB to UART converter or by writing a custom firmware that will remove the device id from flash.

AlexHighTower, 2020-09-30:
I also have related problem...
While there is no custom fw for MHO-C401 I decided to listen to original advertisement data from device my esp32
I fond that data is encrypted and key is needed and it can be received from WebFlasher. WebFlasher provide it to me, but if I press "activate" button several times, I'll get different keys and tokens... Is is normal?

Another problem I found is that device send each 5 seconds the same data, in my case 09094d484f2d43343031 - it doesn't looks like valid data and I have no idea what is it...

I get only ones big packages when esp2 starts to scan ble devices... they looks like:
0201060f1695fe305887035f3ec45438c1a40809094d484f2d43343031
0201061a1695fe58588703603ec45438c1a4a940bec5710100002d059a0409094d484f2d43343031

According to [HAAS](https://github.com/custom-components/sensor.mitemp_bt/blob/31f83bdf3d1454074246375ab57c32ca25584e5b/custom_components/mitemp_bt/sensor.py#L205) sources first package is NOT encrypted and the second one does...

what we have for second one
020106 - BR/EDR + LE General discoverable mode flags
1695fe - Xiaomi service data
8703 -product id
60 - index / counter
3ec45438c1a4 -reversed mac (original is a4:c1:38:54:c4:3e, reported by esp32 and WebFlasher)
5858 - frame control
8703 - sensor type
last 31 - rssi
framectrl & 0x0800 != 0 -> encrypted

try to decode
data to work with = a940bec5710100002d059a0409094d484f2d433430
aad =  11
token =  2d433430
payload_counter =  4d484f
nonce =  3ec45438c1a48703604d484f
cipherpayload =  a940bec5710100002d059a040909

key from WebFlasher = b95817fcf572e462ea80d5f9e99889bf

and I try in python 
        cipher = AES.new(key, AES.MODE_CCM, nonce=nonce, mac_len=4)
        cipher.update(aad)
        plaindata = cipher.decrypt_and_verify(cipherpayload, token)
and fails
ValueError: MAC check failed

first not crypted I can't parse too... as well as strange 09094d484f2d43343031

any Ideas about that?

nikito7, 2020-09-30:
Cypher is smaller I think

Its about ID

Web flash "kills" mho

nikito7, 2020-09-30:
You can flash custom, but lcd will not work

AlexHighTower, 2020-09-30:
device itself continue to work properly and show correct data
may be I parse package in wrong way... I have no idea, actually...
and I see link to Magalex2x14/LYWSD03MMC-info, unfortunately I can't see anything similar here (((

nikito7, 2020-09-30:
Device ID is wrong now, no solution.

For others: dont use activate on MHO!!!!!!

atc1441, 2020-09-30:
I think i will make a firmware that erases the ID section to restore default values, 

on an already activated device it should work to activate it in the webflasher.
then it will not set a custom one, is is needed to be set so there is better solution than to set a custom one.

Also the key will be new on every activation as this is what the activation does, it generates new secure keys :)

atc1441, 2020-09-30:
I just tested to activate an already activated MHO-C401 in the webflasher and then reactivated it in the Mi home app without any problems.

It is of course needed to delete the device in the mi home app first and then add it again after an activation in the webflasher but if one time activated in MiHome app there is no problem

nikito7, 2020-09-30:
I never activated in MiHome, only web flasher

Yvon-Indel, 2020-10-01:
Hi, I guess I'm in the same situation.
I used a new sensor just out from the box. Connect in the web flasher, do activation, uploaded custom firmware. No error message during all the process.

I can change the parameters through the web flasher...seems to be ok.

But the sensor do not advertise anymore. I can see data from my others original firmware sensor.
I try changing advertising delay with no luck
I try to put the stock firmware back, but now I can't bind the sensor with  Xiaomi home application. Same error as above.

Did you have an Idea  why the sensor is not advertising ?

Thanks a lot.

atc1441, 2020-10-02:
Hey. Is it an MHO-C401 or the LCS version ?

Yvon-Indel, 2020-10-02:
Hi,

This is the LCS version, this one : 

[image]

atc1441, 2020-10-02:
Ok so it is not the version talked about here. 

It looks to me like Xiaomi blocked the custom IDs so there is definitely a firmware needed that will erase the id on flashing. Could be even implemented in the current custom firmware. So if you flash it once and go back it will create a new id.

Yvon-Indel, 2020-10-02:
Ok sorry.
Could I use it with your custom firmware ? Because the trouble is the sensor is not advertising...can't see any data.

atc1441, 2020-10-02:
Yes that should be no problem.

Try to reflash the custom firmware and see again.

You can also check with nrfconnect if the data is there and if the reason it the thermometer or the other end

atc1441, 2020-10-02:
The newest update pushed right now will erase the memory section where the id is saved :)

Now lets hope that they dont do an aditional MAC blocking, if so we can also erase that section but then the device will get a new MAC

nikito7, 2020-10-02:
Fixed :-)

nikito7, 2020-10-03:
@atc1441 

Can you check if decrypting code is the same as LYWSD03MMC. 

Because not working in esphome

atc1441, 2020-10-03:
It should be the same as the activation and flashing works just like that.

Will check it further when coding the custom firmware for it but it will take time

nikito7, 2020-10-03:
Maybe advertising code changed

Yvon-Indel, 2020-10-03:
> Yes that should be no problem.
> 
> Try to reflash the custom firmware and see again.
> 
> You can also check with nrfconnect if the data is there and if the reason it the thermometer or the other end

Hi,
Ok  the device is advertising again. 

Thanks

michapr, 2020-10-04:
Hi,
I'm a bit confused... ;)
MHO-C401 is now compatible with latest custom firmware? 
Do not want to try it because not sure how I can go back to original firmware for MHO-C401 (from where get it...)

Thanks!
Michael

atc1441, 2020-10-04:
No its not compatible for now. 

Ypu can flash the custom firmware but only ble and not the display will work but you can flash back the stock firmware without a problem.

The custom firmware would be only needed right now to to reverse the Id that Xiaomi blocked after activation with the web flasher

closed 2020-11-12 by atc1441