atc1441/ATC_MiThermometer #25: TypeError: device.watchAdvertisements is not a function

rkoebrugge, 2020-09-26:
I received my Xiaomi sensors today. I flashed the first device with help of the YouTube video. After flashing the device itself presented its MAC address, but when I connect from the TelinkFlasher the log said "TypeError: device.watchAdvertisements is not a function". After that it will try to reconnect. It also detects the custom firmware. The flasher page does not show any Temp/Humi data anymore

Did i made a mistake? I tried flashing it again, same procedure with the same result.

Also, my OpenMQTTGateway ESP32 doesn't pick up a temperature and humidity. It does get detected with
{"id":"a4:c1:38:90:29:62","rssi":-85,"distance":15.12852,"servicedata":"a4c13890296201052c570bb001","servicedatauuid":"0000181a-0000-1000-8000-00805f9b34fb"}

atc1441, 2020-09-26:
Sounds good so far. 

The Sensor data will not be shown in the web flasher on custom firmware. 
So if it says custom firmware detected everything is fine

rkoebrugge, 2020-09-26:
Thans for your quick answer!
Ohh, I'm sorry. I expected the custom firmware also be able to send the Temp/Humi.

Do you perhaps know what to do to get the sensor values available in OpenMQTTGateway?

atc1441, 2020-09-26:
No problem. 
But unfortunately i dont use MQTTGateway

The servicedata you received is allready the vorrect data.
See the readme for an explanation 
https://github.com/atc1441/ATC_MiThermometer#advertising-format-of-the-custom-firmware

rkoebrugge, 2020-09-26:
I'll try updating OMG first, I see a 0.9.5 version where it should be included and I probably use an older version.

closed 2020-09-26 by rkoebrugge