atc1441/ATC_MiThermometer #26: Stuck on "Detected custom firmware"

omerbitas, 2020-09-30:
Hi, thanks for the great work.

I installed the custom firmware, it shows correctly on the device, when i try to connect to it, i get this:

Log:
00:51:04: Searching for devices
00:51:14: TypeError: device.watchAdvertisements is not a function
00:51:14: Reconnect 1 from 5
00:51:18: Detected custom Firmware

And that's it. No temp or humidity indications or other.
Any idea what can cause this?

atc1441, 2020-10-01:
Hey.
That is exactly what should happen so the flashing was correct

omerbitas, 2020-10-01:
Thanks for the response.
ok so the flash was successful,  yet I cannot see the device in HA using mitemp_bt
do you have any experience with integrating the sensor to HA? the config is:

>   - platform: mitemp_bt
    whitelist:
      - 'A4:C1:38:7B:BB:B3'

closed 2020-10-01 by omerbitas

atc1441, 2020-10-01:
Unfortunately I can not really help you with that

 I don't use it myself