atc1441/ATC_MiThermometer #28: Firmware changelog

nchieffo, 2020-10-05:
I noticed that the .bin has been updated since my first installation last week.
Is it possible to create a firmware changelog so I can understand what are the new bufixes and features?

atc1441, 2020-10-05:
Oh you are right. 

So far the updates are descripted in the differend issues but for a non involved person its hard to follow. 

Will start to make a changelog on new updates. 

The latest update was a bugfix on the lcd turning of and the firmware will now remove the Mi id of the device from flash as Xiaomi was blocking the one who had the custom firmware installed and then where flashed back to stock but thats no problen anymore

m-kozlowski, 2020-10-05:
@atc1441 I don't want to meddle in the way you work, but proper commit messages would make it easier to track changes, and would act as "self documenting" changelog;)

atc1441, 2020-10-11:
Will use the commit messages in future :)

closed 2020-10-11 by atc1441