atc1441/ATC_MiThermometer #29: blinking smiley

michapr, 2020-10-05:
Hi,

thanks for the very good job!
Sensors are nice, but battery life time is short if make a request every 10 minutes....
So I hope will be much better with advertising only.

regarding the blinking smiley - is it possible to go back to the default setting, that smiley is permanent available (even by default)?
Maybe same for battery could be useful.

Or maybe create a second such firmware file? ;)

Thanks!
Michael

atc1441, 2020-10-05:
You can only change it via the webflasher. 

I am planing to do persistense storage of the settings but it will take time to implement it.

I will not make extra firmware files for other settings as its to mich hassle to update everyone when a new update comes out

m-kozlowski, 2020-10-05:
@michapr It took me ~5 minutes to get build environment up and running.
Grab a [toolchain](http://shyboy.oss-cn-shenzhen.aliyuncs.com/readonly/tc32_gcc_v2.0.tar.bz2), [SDK](https://github.com/Ai-Thinker-Open/Telink_825X_SDK), export PATH to toolchain's binaries, put proper path to SDK in ATC_Thermometer/makefile and you're ready to change any defaults you need;)
Most of the settings are at the top of app.c file.

michapr, 2020-10-05:
> It took me ~5 minutes to get build environment up and running.

Thanks, you are right,.... took about 5 minutes after I have found a 64bit LINUX system here ;)
(all my Raspberrys are under 32bit... and under Windows don't like to do it.)

Have compiled the current build - will make my modifications then.

Thanks again!
Michael

closed 2020-10-05 by michapr