atc1441/ATC_MiThermometer #32: Can't connect to newly flashed device

jbaiter, 2020-10-06:
I just successfully flashed a bunch of thermometers with the custom firmware, but now I can't connect to them with the web tool to update the settings:

```
Log:
21:39:09: Searching for devices
21:39:18: Connecting to: ATC_8A6105
21:39:28: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
21:39:28: Reconnect 1 from 5
21:39:28: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
21:39:28: Reconnect 2 from 5
21:39:28: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
21:39:28: Reconnect 3 from 5
21:39:28: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
21:39:28: Reconnect 4 from 5
21:39:28: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
21:39:28: Reconnect 5 from 5
21:39:28: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
21:39:28: Something went wrong, to many reconnect's
```

They seem to be working fine otherwise, switching between displaying temperature/humidity and temperature/battery and advertising their values over BLE.
Is there another way to update the settings, besides the web tool?

atc1441, 2020-10-06:
Please try to restart the pc and the devices and see if it helps.

Alsp please try it with an Android smartphone if possible

jbaiter, 2020-10-06:
Thanks for the hint and the incredibly quick response, works perfectly from my phone :-)

closed 2020-10-06 by jbaiter