atc1441/ATC_MiThermometer #33: Advertising interval

michapr, 2020-10-07:
Hi,

how to understand the "Advertising interval:"?

I have changed it via web flasher, but cannot see a difference - advertising values will be received every minute about.

Then have changed in code (app.c):
`RAM uint8_t advertising_interval = 6;//multiply by 10 for value`

to

`RAM uint8_t advertising_interval = 36;//multiply by 10 for value`

but no difference too (other values like smiley, battery will be accepted in code and web flasher as expected)

I want to change the enable sensor time for saving battery, as sample to 3 minutes.

Thanks!
Michael

m-kozlowski, 2020-10-07:
with 36 multiplier you're running into integer overflow.
`if((clock_time()-last_adv_delay) > advertising_interval*(advertising_type?5000:10000)*CLOCK_SYS_CLOCK_1MS)`
so it'ss: 36*10000*24000 = 8640000000, twice as much as uint32 can take;)

atc1441, 2020-10-07:
Correct.

This is a future fix. Need to get myself into it :)

michapr, 2020-10-07:
`with 36 multiplier you're running into integer overflow.`

ok, that's why the 5 and 10 minutes from web flasher cannot work too, right?

For understanding - what time interval is it?
I still see advertising data every minute, or I'm wrong? Is it the interval for sending an *actual* value in advertising packet?

Thanks for help!
Michael

m-kozlowski, 2020-10-07:
@atc1441 skimming through docs i've found `clock_time_exceed( unsigned int ref, unsigned int us )` that handles the whole clock frequency part, taking interval in microseconds as uint. that would give max interval of 4294 seconds without introducing second variable to count interval ticks

atc1441, 2020-10-07:
Thank you for that. 

I did used that method at the beginning but had the same problems with the overflow. Will look into it again

m-kozlowski, 2020-10-07:
On the second thought, there will be still problem of internal clock overlaping somewhere between 100-200seconds, depending of clock frequency, no matter which function is used to measure time.

atc1441, 2020-10-07:
Thats what i was also seeing when using it. 

A plan is to have an extra Arduino like millis timer to have independent precise timing available. 

So ((overflow counter)*X) + (current counter value) = millis