atc1441/ATC_MiThermometer #34: suggestion: ability to turn off LCD (save battery)

gurkburk76, 2020-10-09:
Would it be possible to add an option to turn off the LCD in the firmware?
Since many of us have these devices where there are never seen it could extend their battery life 👍

atc1441, 2020-10-09:
A problem is that we dont know how to turn the screen of.

There are no infos available about the controller and we only have i2c and no reset pin.

What would be possible is to turn off updates of the LCD

ralf-e, 2020-10-10:
is there read access to the LCD controller? If not, unsolder ...

atc1441, 2020-10-10:
The stock firmware does not read from the controller. 
So we dont know how to read.

The stock firmware only writes the init and then the segment sequenz.

We could simply only send the init and never really update its just the question how much it saves

m-kozlowski, 2020-10-10:
segment lcd current draw is low. Very low. Probably somewhere around 1-2uA. Typical cr2032 is enough to run it for more than 5 years.
Comparing it to the power consumption of mcu, bluetooth communication and sensors it's absolutely negligible.

atc1441, 2020-10-10:
Yeah BLE is definitely the most consumption

closed 2020-11-12 by atc1441