atc1441/ATC_MiThermometer #35: Scanner doesn't show any devices

animal6496, 2020-10-10:
I got very interested in this project as I have 4 of these Xiaomi devices and would like to integrate them in my home automation system. So I downloaded the firmware and used the Chrome app to got the TelinkFlasher page and started scanning and then nothing... The log says Searching Devices. I read that this can take some time but after half an hour I gave up. I checked with a BLE scanner on my iPhone and all 4 devices show up. I then used my iPad and laptop (running Ubuntu 20.04), results are consistent, nothing happens.  I noticed in the video that hitting the Connect button opens a Bluetooth connection screen but that doesn't happen in my case. On iOS devices Chrome doesn't throw a Bluetooth access dialog and I can't find a way to do this manually  in settings. Can someone enlighten me how to set this up on iOS 13 (or Ubuntu) or is this Android only?

atc1441, 2020-10-10:
Please take a look at this. 

https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md

That will help you find compatible devices

animal6496, 2020-10-10:
OK that explains a lot. On iOS I do not immediately see a solution, may be Opera works if I can get the flags set.
Linux route looks more promising. I used the flags for Chrome on Linux now it does scan but refuses to  connect need to check on BlueZ as suggested.

animal6496, 2020-10-10:
For Ubuntu 20.04 BlueZ is default correct version and active. Just a reboot was enough to get everything working from my laptop and I already flashed the first device. Thanks for you quick answer.

atc1441, 2020-10-10:
Great

closed 2020-10-11 by atc1441