atc1441/ATC_MiThermometer #37: Make 0xAF = Mi Like default

kiwipaulrob, 2020-10-10:
I am setting these sensors up in home assistant using the blue HACS add in. It expects the sensors to broadcast using the Mi Like format. I can change the sensors to transmit using the Mi Like format but this will reset to the custom format when I change the battery. Can you consider making the Mi Like format the default? Thank you for your excellent work.

atc1441, 2020-10-10:
Hey. 

I will not do that.

You can compile either your own firmware or in future i want to make the settings persistense but that will take time

kiwipaulrob, 2020-10-11:
I understand and that is reasonable. Could you possibly generate a bin with Mi Like as the default?

michapr, 2020-10-12:
I compiled for you:
https://smarthome-work.com/download/ATC_Thermometer_MiDefault.bin
Take it for your own risk - I will remove it soon because will not be updated if any changes in app.
Hope it will help.
@atc1441 - I hope you agree with it.

kiwipaulrob, 2020-10-13:
Thank you that is very kind. Best wishes.

juhjar, 2020-10-31:
> I compiled for you:
> https://smarthome-work.com/download/ATC_Thermometer_MiDefault.bin
> Take it for your own risk - I will remove it soon because will not be updated if any changes in app.
> Hope it will help.
> @atc1441 - I hope you agree with it.

Cool, as you have build environment setup, could you build following: MiLike, no smiley, battery level disabled, 5min interval.
Thank You...

michapr, 2020-11-01:
Hi,
@juhjar - here is your version ;)
https://smarthome-work.com/download/ATC_Thermometer_MiDefault_noSmiley_150s.bin

Because of this I have used 150 seconds advertising interval:
https://github.com/atc1441/ATC_MiThermometer/issues/33

Soni93200, 2020-11-02:
> Hi,
> @juhjar - here is your version ;)
> https://smarthome-work.com/download/ATC_Thermometer_MiDefault_noSmiley_150s.bin
> 
> Because of this I have used 150 seconds advertising interval:
> #33

Thanks for this, solved my problem, sensor was not detected with original custom firmware :p

benek984, 2020-11-03:
@michapr If this is too much just ignore but could you compile for me as fallow: Smile: Comfort indicator, MiLike, Battery: Disabled, Advertise: 10 seconds.

Thank you in advance.

atc1441, 2020-11-03:
We definitely need an online compiler for this :D

TheNitek, 2020-11-03:
Or persistent settings storage ;-)

atc1441, 2020-11-03:
Yeah that would solve a lot as well maybe ^^

michapr, 2020-11-04:
@benek984 - your version ;)
https://smarthome-work.com/download/ATC_Thermometer__benek984.bin

closed 2020-11-12 by atc1441

attley86, 2020-12-20:
Issue was closed  #33 but I would appreciate if someone could compile a binary for me that has: MiLike, no smiley, battery level disabled, 5min interval. Cheers.