atc1441/ATC_MiThermometer #38: accidently flashed the zip, not the bin

gbomacfly, 2020-10-10:
Hi there,

thanks for this tool. But I made a mistake. I flashed the zip-file, not the extracted bin.
I'm trying to flash through UART through a USB-Adapter. It seems to write and finish, but the device is still dead.
I've tried with RX and TX, but both with no luck. Also the stock- and your .bin
Some other ideas?

atc1441, 2020-10-11:
If you are on mac please try it with a windows mashine. Flashing works bettere there

atc1441, 2020-10-11:
Added a real file check to the webflasher so its only possible to select a real firmware file for telink.

This does not help you but in the future

gbomacfly, 2020-10-11:
> If you are on mac please try it with a windows mashine. Flashing works bettere there

Thanks a lot, this had worked fine :)
And also thanks for the filecheck :)

closed 2020-10-11 by gbomacfly