atc1441/ATC_MiThermometer #42: Two devices, two different TX/RX behavior

glanch, 2020-10-13:
First of all, thanks for your awesome work - this is amazing.
I have two LYWSD03MMC sensors - both flashed with your provided .bin build in `master` branch. I am using it in combination with the OpenMQTTBridge on an ESP32. One sensor is behaving perfectly fine. The other one only works if it is placed within a circle with an radius of approximately 1.5m centered at the ESP32. I tried flashing the sensor over and over again, rebooted it by taking out the battery, replaced the battery, switched the batteries among both devices, turning one device off while the other one is on, placing them in the same room, placing them in different rooms with the same distance to the ESP32 - it makes no difference. Does anyone have a clue what the problem is?

atc1441, 2020-10-13:
Try to reflash the stock firmware and see if the reception is better

CWGSM3V0, 2020-10-23:
@glanch Did this end up working for you? I have 9 of these LYWSD03MMC units and only one seems to have an issue with range/advertising interval (seems to advertise whenever it feels like it, sometimes several hours apart vs the other units.) It did this with the stock firmware as well. Not sure if the unit is damaged or not but it flashed ATC without issue and so far has only advertised once. If I connect via NRF Connect, I can see the data real-time. Also using OpenMQTTGateway.

CWGSM3V0, 2020-10-23:
Flashed remaining 8 units and all work exceptionally well.

I'd say much better than the stock firmware as OpenMQTTGateway is showing updates ~60secs to 4min max (edge case) whereas in stock it was much less predictable due to how OMG had to work around encryption. For my use, that's acceptable and based entirely on the BLE power of my M5StickC (as I used my laptop to flash them OTA and I had no issues reaching even the -93dbm one and all the flashes took around 60 seconds) and I'll probably upgrade to the same unit as the author of OMG recommends.

The troublesome unit only works 1-2 feet away from the ESP32/OMG whereas the rest are scattered around my house and work fine. I'm going to call it a day and say it's a defective unit (I had issues with range even with stock firmware; I had to literally be in front of it with my phone running the MiHome app) and replace it for $7 shipped from AliExpress.

@atc1441 Love the work and effort you put into this. One year is 100% dedication we all have benefited from. I honestly believe your work makes these little devices much better than they were out of the box.

mr-sneezy, 2020-10-25:
I'm waiting for one of these units to ship to me so no direct experience with them yet, but it might be worth checking the PCB for poor component soldering around the BLE antenna or in fact a missing component before throwing it out. A range of two feet is about right if the antenna is not connected...

CWGSM3V0, 2020-10-25:
> I'm waiting for one of these units to ship to me so no direct experience with them yet, but it might be worth checking the PCB for poor component soldering around the BLE antenna or in fact a [...]

I'll check that out, never even occurred to me. Thanks 👍

glanch, 2020-10-26:
> I'm waiting for one of these units to ship to me so no direct experience with them yet, but it might be worth checking the PCB for poor component soldering around the BLE antenna or in fact a [...]

I checked the PCB and tested some components for continuity. Both PCBs (of good and bad units) look the same and behave the same continuity-wise.

mr-sneezy, 2020-10-27:
Just saw the edit. Maybe there was a bit of copper or solder floating around that you dislodged.

CWGSM3V0, 2020-11-01:
So I ended up buying a Olimex ESP32-GATEWAY-EA as per OMG documentation and I have to say, this little unit is amazing (bought the box for it too) and it can detect and read the flaky sensor (read: short-short-range) I have. I guess even if the sensor is bad, throwing enough hardware at the other end of the equation saves me $7 :)

mr-sneezy, 2020-11-02:
My first unit arrived and flashed fine and has a range of over 10m and through a wall, so I bought three more 👍

closed 2020-11-12 by atc1441