atc1441/ATC_MiThermometer #43: Measure battery more seldom

orrpan, 2020-10-14:
As I understand measuring the battery is the thing that drains it the most, is it possible to only do these measurements like once a day or at least once a hour?

atc1441, 2020-10-14:
Hey.  that would be my plan to do but right now nothing can be more than 2-3 minutes apart because of the overflow Bug in the future there will be real working millis counter

tshcherban, 2020-10-16:
As far as i can see from sources, currently battery is measured once per 5 minutes
[image]
@orrpan, have you tried to adjust `Read battery delay`? I've successfully changed it to 15 minutes.
@atc1441 where the overflow bug is happening? Can you provide a bit more details?

TheNitek, 2020-11-01:
Wouldn't it be possible to disable battery measurements completely? Personaly I don't care about the value, if the device is dead I'll just replace the battery, but I like the idea replacing the battery more seldom ;-)

atc1441, 2020-11-01:
Thats a good argument!

It can be simply disabled by removing the battery part in the app main loop

closed 2020-11-12 by atc1441