atc1441/ATC_MiThermometer #45: Parameters non-volatile storage

tshcherban, 2020-10-16:
Would be nice to store offsets, advertising type, smiley preference etc during battery replacements.
I've failed to find flash addressing, so tried from **_tiny_Flash_** lib example:
`tinyFlash_Init(0x70000,0x4000);` in `main`, then
`tinyFlash_Write(1, &temp_offset, 1); tinyFlash_Write(2, &humi_offset, 1);` in `cmd_parser` and
`tinyFlash_Read(1, &temp_offset, 1); tinyFlash_Read(2, &humi_offset, 1);` in `void user_init_normal(void)`
but now device is bricked (stuck at startup screen), will try to flash it via USB-Serial method.

atc1441, 2020-10-16:
So far i only worked with direct access to the flash as you can see at the Mi ID erasing. Will also do the non volatile storage without tinyflash and with direct flash access and a custom driver

nalt, 2020-12-02:
Would be quite useful. For instance: In cold environments, the battery runs empty more quickly. or the device shuts down temporarily, restarting with default settings

atc1441, 2020-12-02:
Yes no question on that, its just a lot of work that i currently have not much time for :)

anacruza, 2021-01-06:
Hi, thanks for your incredible software... I'm following the new releases, but maybe I missed it. Is there any improvements regarding this issue? Regards!

atc1441, 2021-01-06:
Hey. No right now its more worked on the tools around the firmware and not the firmware itself.

The non volatile storage is still untouched.

atc1441, 2021-01-06:
Since @pvvx made some very nice features into the Firmware i suggest on checking out his version:
https://github.com/pvvx/ATC_MiThermometer

this includes non-volatile storage and a better low power management

rvben, 2021-01-06:
@atc1441, are you considering pulling in (some) of the improvements that have been implemented by pvvx, or what is your view at this moment?

atc1441, 2021-01-06:
if i pull it now it would basically mean i copy his complete Repo, as nearly every file changed, i can really suggest on sticking to his repo for now :)

got just to less time right now to get deep into the Thermometer again