atc1441/ATC_MiThermometer #47: Help with Sensor instant Advertising

nchieffo, 2020-10-18:
I pushed the FC01 value to my device, but I can't see any difference when reading values from home assistant.
The value seems to update every 5 minutes even if there is 0.1° change in temperature
Can you help me?

atc1441, 2020-10-18:
It is possible that the home assistant is only allows reading every 5 minutes. Maybe you can set it there

nchieffo, 2020-10-18:
Hello and thanks for the reply

Actually the period is 60 seconds
This is the mod I'm using and the option:
https://github.com/custom-components/sensor.mitemp_bt#period

Is there any way to detect sensor advertising using Android app, just to see if the sensor is correctly advertising every 0.1°C?

atc1441, 2020-10-18:
You can use nrfconnect app on android and will see the changing packet id(last byte) if the advertising is correct

nchieffo, 2020-10-18:
Can you also suggest a command line util (also linux ok) to capture adv and read values (using Xiaomi like advertising)? With nrfconnect I have problems understanding how to use it, sorry

atc1441, 2020-10-18:
I dont know any. Sorry.

With nrfconnect you can see how i use it in the video

closed 2020-11-12 by atc1441