atc1441/ATC_MiThermometer #48: Automatic smiley changing like on stock firmware

Asica10, 2020-10-18:
Would be nice to add automatic changing smileys like in stock firmware in reference to temperature and humidity. 
Right now default option with blinking smiley require stay few second and wait when smile will appear.
Thank you in advance.

atc1441, 2020-10-18:
The smiley in the custom firmware has not real function. 

It is just there to see that the firmware is running. You can set the smiley via the web flasher to be allways happy or sad or of but i gavent add any corelation to the temp/humi. As i dont like the finction myself, everyone would hase different expressions of the right smiley :)

Asica10, 2020-10-18:
Ok understand, I do not pay attention on smil:)  but it is helpful for my wife, to see  that temp and humidity are ok or not.

hkapanen, 2020-10-22:
Would be great to have programmable temp-hum region for the smiley.

popsodav, 2020-10-25:
I've created pull requests for this feature:
https://github.com/atc1441/ATC_MiThermometer/pull/53
https://github.com/atc1441/atc1441.github.io/pull/5

I've used this document to detect comfort zone for temperature and humidity:
https://www.researchgate.net/publication/333189109_Comparison_of_Defuzzification_Methods_in_Case_of_Air_Conditioning_Systems

closed 2020-11-12 by atc1441