atc1441/ATC_MiThermometer #49: What is the differences?

zuidwijk, 2020-10-18:
Can you tell me what the differences are between these two screens?
[image]
[image]
Both got the same temperature. The display without the smiley is/should be humidity. What is the other?

atc1441, 2020-10-18:
Hey. The smiley itself is just blinking. 

The one with battery symbol is showing the battery in percent at the humidity data field. 

The one without is just humidity

zuidwijk, 2020-10-18:
Thought so! Thank you for confirming :)

closed 2020-10-18 by zuidwijk