atc1441/ATC_MiThermometer #50: Differences in humidity reading

Asica10, 2020-10-19:
I would like to report that after flashing custom firmware humidity reading are 4-5 % lower than on stock firmware. I tested it on 3 pcs and each had this issue.
Is this intentional behaviour or bug?

atc1441, 2020-10-19:
Hi  that is known.

The values of the custom firmware are the ones directly from the sensor, it seems that the stock firmware does add some offset to it.
I added a function to the firmware so that you can set a custom offset if wanted

closed 2020-11-02 by atc1441

tasosmos, 2021-04-26:
Hi, I noticed this issue as well and did a quick comparison between 5 Mi devices and 2 official Sensirion SHT31 "smart gadget" devices (higher precision sensors and presumably reliable firmware).

In most cases the stock firmware humidity measurements are closer to the "reference" sensors.

e.g. 
Reference: 21.2 C, 59.5% RH - 21.37 C 59.9% RH
Stock firmware: 21.2-21.3 C 61-62% RH
Custom firmware: 21.5-21.7 C 54-56% RH