atc1441/ATC_MiThermometer #51: Device security - disable OTA, custom firmware settings

bangom, 2020-10-22:
Hello,

thank you for the great project!

Do I understand correctly, that the MiThermometers with original or custom FW can be flashed (or bricked) via OTA by anybody within the BTLE range? Same goes for the "Custom firmware Settings" (temp / humidity offset...)?

Is there any option to disable the OTA / Custom firmware setting? For example after 1min after inserting the battery or by HW modification to the MiThermometer (shorting some  "program-disable" pin on the PCB)?

Lastly, is there an option how to prevent spoofing of the MiThermometers data? For example by implementing encryption/signing data with bindkey, so we can verify validity of received data?

Just trying to have secure home automation and credible environmental data for central heating control.

atc1441, 2020-10-22:
You are right with the OTA update
A more secure version could be implemented but is not right now.

When you use the stock firmware with the encryption key it should be spoofproof. Right now the custom firmware does not use any encryption

bangom, 2020-10-22:
> You are right with the OTA update
> A more secure version could be implemented but is not right now.
> 
> When you use the stock firmware with the encryption key it should be spoofproof. Right now the custom firmware does not use any encryption

But because anybody in the BTLE range can Activate the device and reset Token, even original Firmware is not hackproof? Because you are Telink expert... is there a HW option to disable OTA updates? I really don't like the idea that any thermometer is open to OTA updates / reactivation by anybody...

ejalal, 2020-10-22:
Good spot @bangom, but this is the case for the original firmware too. You should go complain at Xiaomi for not securing the original firmware first but if they listen it will not be possible to hack them like @atc1441 did brilliantly.

atc1441, 2020-10-22:
It is very simple to deactivate the ota function. It is also possible to do something like a password you can set before the ota update is working. See the ota part in the firmware.

The stock firmware is spoof proog with the encryption because if someone would repair the device and "spoof" something the encryption key would change you and the previous encryption does not work anymore and your will not get false positives. It is of course still possible to load a new firmware on to it and brake it but not in a spoof way, only talking about stock firmware here

wwwouter, 2020-11-07:
I would love to have an option where I can set a password, and then afterwards every action will need that password, or it will fail.