atc1441/ATC_MiThermometer #52: Feature Request - Show BT MAC Address of Sensor in Web Flasher

Teleportist, 2020-10-23:
I find it quite a lengthy process to identify the MAC address, when trying to add a new sensor to ESPHome.

Mu mobile is an iPhone and lamptop is running Windows 10. Neither of which seem to make it easy to ID the
sensor.

Would you kindly add a box to display the MAC address of the connected device in the Telink Flasher please?

atc1441, 2020-10-23:
Unfortunately it is not possible just like that  the web bluetooth api has no way of getting the mac of the connected device. Allready tried to implement that feature. 
But when more than one device with the same name is available it will show the mac in the search window behind the name

nalt, 2020-11-07:
Then it would be helpful to show the device name. Since you have the MAC in the name with the custom firmware, you could distinguish devices with custom firmware. In some cases, there is a console log that shows the name.

atc1441, 2020-11-12:
Yes the device name is shown in the log allready so you know to what device you are connected to

Teleportist, 2020-11-12:
The problem is I have 6 sensors and all the device names are the same!

atc1441, 2020-11-12:
Yes on stock firmware all have the same name,

The best way now would be to only insert the battery in one at a time, or look after connecting to the device which one has the ble symbol on

closed 2020-12-07 by atc1441