atc1441/ATC_MiThermometer #55: Can't connect on Android 4.4.2

vdiogo, 2020-10-25:
Since I can't use the flasher on iOS I picked up a old tablet with Android 4.4.2 where I installed nRF Connect. nRF Connect can connect to the Xiaomi device but can't seem to connect using the flasher (tried several combinations): when pressing the button Connect I get a "Searching for devices" + "NotFoundError: Bluetooth Low Energy not available" + "Reconnect 1 from 5" messages and nothing happens.

KoalaBear84, 2020-10-25:
"NotFoundError: **Bluetooth Low Energy not available**"

The Xiaomi sensors use BLE which is not supported by the tablet. [BLE is supported](https://developer.android.com/guide/topics/connectivity/bluetooth-le) with Android 4.3+

vdiogo, 2020-10-27:
> "NotFoundError: **Bluetooth Low Energy not available**"
> 
> The Xiaomi sensors use BLE which is not supported by the tablet. [BLE is supported](https://developer.android.com/guide/topics/connectivity/bluetooth-le) with Android 4.3+

@KoalaBear84 my tablet has Android 4.4.2 (so over 4.3) and nRF Connect can detect the all the Xiaomi BLE devices so not sure my issue was clear?

KoalaBear84, 2020-10-27:
Indeed, didn't correctly read it and didn't have a tablet model to verify. Included BLE page to check that the Android version isn't the problem.

So then it should be able to work. Which browser do you use? Looks like it should already work even with Chrome 70 which was released on October 16, 2018.

vdiogo, 2021-01-09:
> Indeed, didn't correctly read it and didn't have a tablet model to verify. Included BLE page to check that the Android version isn't the problem.
>
>
>
> So then it should be able to work. Which browser [...]

Apologies for late reply. I've tried with native browser and Chrome

Dixit22, 2021-01-16:
I have exactly the same problem.  I am running Android 5.0.1 on a Samsung Galaxy S4 (this is the latest version of Android the phone supports).  BLE Checker confirms BLE is supported; nRF Connect can connect to the devices with no problem. But the flasher, using Chrome 87.0.4280.141, can not.  
On the other hand I also have access to a Motorola/Lenovo Moto G4 Play (which supports BLE 4.1) running Android 7.1.1 and exactly the same version of Chrome (87.0.4280.141).  Using this I was able to flash a LYWSD03.

mr-sneezy, 2021-01-17:
Slightly OT but the Samsung S4 can go to at least Android 9 if you want to badly enough. I've got a useful extended life out of mine by installing LineageOS.
https://wiki.lineageos.org/devices/