atc1441/ATC_MiThermometer #56: Cannot see UUID 0x181A using ATC_Thermometer.bin from commit 772348a

ADBeasley, 2020-10-26:
I flashed a new unit using the ATC_Thermometer.bin from commit 772348a and it does not match the documentation or report correct values:

My BLE scanners (LightBlue on the Mac and BT Inspector on the iPad) find and connect to the device fine.  Connections drop on both devices after 4-5 minutes.

I can collect the battery status via UUID 0x2A19:
0x2A19 (one byte) -> battery percentage e.g. 0x59 = 81% charge 
**Note on the display the icon shows a nearly flat battery for this charge**

I get two UUID for environmental sensing:
0x2A1F (two bytes) -> High Byte is temperature in 1/10ths ºC in hex e.g. 0xCD00 = 205 = 20.5ºC
0x2A6F (two bytes) -> Possibly data for humidity e.g. 55% humidity = 0x7C15, 56% = 0xE015 and 75% = 0x4C1D
**I've not been able to work out how this conversation works.**

UUID of 00010203-0405-0607-0809-0A0B0C0D1912 that replies 0x00
OTA - UUID 00010203-0405-0607-0809-0A0B0C0D2B12
**No idea what this is**

UUID: 1F10
0x1F1F -> This allows me to update the settings as per the documentation

UUID: FE95
Does not report anything

Please note I used the file called **ATC_Thermometer.bin** as I could not see a file called ATC_Thermometer_custom.bin in the master branch.

Flash was done with Edge on Catalina (OTA using your web page) and the device reset automatically.  Pulling the battery leaves the firmware flashed with the bespoke code fine and the manually applied settings (e.g. smiley off / battery off) are lost as expected / noted.

m-kozlowski, 2020-10-26:
> Note on the display the icon shows a nearly flat battery for this charge
There is only one segment for battery icon. It's either on or off. 

>  I've not been able to work out how this conversation works.
divide by 100

ADBeasley, 2020-10-26:
Thank you for those -
The battery I can live with and just turn it off on the display.  The little 'block' is too close to a 'low battery' warning TBH.

I had just dug out a hex calc for the Humidity and converted the 0x7C15 to a 2 byte unsigned and saw it dropped to 5500 at 55% :-) (or blush).

So if its just the documentation being incorrect would I do best to pull that and offer to update?

atc1441, 2020-10-26:
Hey @ADBeasley 

As for the battery icon, everytime it is showing the humidity number field is showing the battery in percent. I know it a bit unfortunate the way it is but the lcd is a bit limited, because of that i also enabled the option to turn battery infos on the lcd of.

The docummentation in the readme is for the Advertising data and not for the UUID read data as you did.

The OTA uuid is as the name implies for the Over The Air Update so to upload a new firmware.

The 0xFE95 uuid is there so that the mi like advertising data get advertised on that uuid to be compatible with EspHome scripts etc. Please look at some other issues for more infos on that

ADBeasley, 2020-10-29:
Thank you for the fast answer - I've finally found an iOS tool that lets me pull the data from the advertisement - think I need to get the Bluetooth programming bible from the library again before I go much further :-)

Till I can get a test box up and running with Python, I've been using an old Jessie Raspberry Pi (my music player TBH) to play with BLE reading and for anyone hunting for ideas, the simplest way of reading the data is:

`gatttool -b xx:xx:xx:xx:xx:xx --char-read --handle=0x000e`

Obviously:

1. Replace the xx with the Mac address
1. Replace the handle with the hex values from my first post for the data you require.

closed 2020-10-29 by ADBeasley