atc1441/ATC_MiThermometer #57: Strange fluctuating value since flashing the custom firmware

thestraycat, 2020-10-27:
Hi! Love the project, congrats on your work.

Im using the small square 2020 version of the sensor and have successfully flashed it using the custom firmware and all steps have gone to plan as described by your tutorial. However now my humidity data keeps fluctuating between 68% and 96% every 5 seconds, it feels like a cached value is stuck somehow? Any idea what's happening here? After the flashing process i chose 'mi like' type and changed the interval to 5mins. Everything else is stock. Pics below to show the 2 screens that it cycles through.

[image]
[image]

Why is it doing this?

atc1441, 2020-10-27:
Hey. That is the battery status in percent. You see it when the battery icon is showing. You could also deactivate that if you want

thestraycat, 2020-10-27:
Thanks for confirming, it wasn't clear to me! Cheers.

closed 2020-11-12 by atc1441