atc1441/ATC_MiThermometer #58: ESP32 BLE and WiFi constraints

marc-gist, 2020-10-30:
I am noticing some issues when using an ESP32 (ESPHome) to receive and broadcast over WiFi (HomeAssistant or MQTT).

After some research, I realized that the ESP32 can't do BLE and WiFi at the same time, and the ESPHome is switching quickly between them.  Which is likely causing my issues, which are:

ESPHome drops from HomeAssistant for a brief time, causing sensors to switch to "unknown" state.
the ESP32 receiver, misses BLE data from the sensors, so isn't updated as much as I'd like (not reliable receiver really).

Anyone have/know of a project where you use two (2) ESP32's, one to handle BLE and one to handle the WiFi?

TheNitek, 2020-10-30:
I am running (my own)  smart home on a esp32 using BLE and wifi at the same time and it works just fine. If you see an issue like this, I don't think it's a hardware one.

KoalaBear84, 2020-10-30:
Also have ESP32, no problems. Never saw any drop. I do get not as much updated, but still running stock firmware. And the ones that are further away get less updates because of the range.

john-arvid, 2020-11-05:
Try a serial debug and see what is happening. I had a problem where the memory was full. 
Needed to remove the AP fallback in esphome wifi settings.

closed 2020-11-12 by atc1441