atc1441/ATC_MiThermometer #59: question: do the custom settings applied with the flashing tool persist when removing the battery?

chris-gibson, 2020-10-31:
I was able to pair and apply the custom firmware file just fine and also apply the settings once it was flashed. After I was done I unpaired from the sensor and removed the battery and put it back in to make sure it was not keeping a connection to anything. I set the device up in the esphome yaml and the esp does see the device but the entities are stuck with an `Unavailable` value in home assistant. I made sure to give it about an hour just to make sure. Then I decided to pair the sensor back up with the flashing tool and re-apply the `Advertising Type` setting and suddenly I started receiving values in the esphome log as well as seeing the values persisted into home assistant.

So do the settings that are applied by the flashing tool persist after the battery is removed or will I have to re-apply the settings when I change batteries?

atc1441, 2020-10-31:
Hey

These values get reset when the battery is reinserted, so you have to set them again.
I want to change that in the future

chris-gibson, 2020-10-31:
Thank you for the quick reply. I will just remember to re-apply them for now.

closed 2020-10-31 by chris-gibson