atc1441/ATC_MiThermometer #60: Read data python

marttin-h, 2020-10-31:
Hello, 
Can you help me. I'm trying to counter the data via a python script in Raspberry, which will be run via a scheduler.
I am unsuccessful in reading data.
Thanks
```
from bluepy import btle
from bluepy.btle import Peripheral

macTest = "A4:C1:38:29:A8:64"

print("Interface: 0 = default")
p = btle.Peripheral(macTest, iface=0)
print("Connect")

services=p.getServices()
for service in services:
    print(service)

data1=p.getServiceByUUID(0x181A)
test=data1.getCharacteristics()[0].read()
humidity=int.from_bytes(test[1:3],byteorder='little')
print(humidity)

p.disconnect()
```

atc1441, 2020-10-31:
Looks good already, what is your problem?

marttin-h, 2020-10-31:
if I read getServiceByUUID (0x181A) I don't get data. Am I doing anything wrong ?

atc1441, 2020-11-02:
Please look into how to get the Advertising data in Python, reading the chars directly is very power hungry and will drain the battery a lot

nadnerb33, 2020-11-08:
Here's something I put together using [py-bluetooth-utils](https://github.com/colin-guyon/py-bluetooth-utils) and the scan example. 
It will print the values and also publish to MQTT
I'm not sure if the way of filtering the advertisements and extracting the values is the best way, but it is working for me.

```
import sys
import bluetooth._bluetooth as bluez
import datetime
import os
import json
from bluetooth_utils import (toggle_device,enable_le_scan, parse_le_advertising_events,
                             disable_le_scan, raw_packet_to_str)
mac_list = [
	"xx:xx:xx:xx:xx:xx",
	"xx:xx:xx:xx:xx:xx",
	"xx:xx:xx:xx:xx:xx",
	"xx:xx:xx:xx:xx:xx"
]

dev_id = 0  # the bluetooth device is hci0
toggle_device(dev_id, True)

try:
	sock = bluez.hci_open_dev(dev_id)
except:
	print("Cannot open bluetooth device %i" % dev_id)
	raise

enable_le_scan(sock, filter_duplicates=True)

try:
	def le_advertise_packet_handler(mac, adv_type, data, rssi):
		data_str = raw_packet_to_str(data)
		if data_str[6:10] == '1a18': #filter out non-atc advertisements
			temp = int(data_str[22:26], 16) / 10
			hum = int(data_str[26:28], 16)
			batt_pct = int(data_str[28:30], 16)
			batt_mv = int(data_str[30:34], 16)
			#print("BLE packet: %s %s %s %d" % (mac, adv_type, data_str, rssi))
			print("%s - MAC:%s  Temp:%sc  Humidity:%s  Batt:%s  Batt:%smV Counter:%s" % (datetime.datetime.now().time(),mac, temp, hum, batt_pct, batt_mv, rssi))
			publish_mqtt(mac, temp, hum, batt_pct, batt_mv)

	def publish_mqtt(mac, temp, hum, batt_pct, batt_mv):
		topic = 'mi_atc/' +  mac.replace(":","")
		message = {}
		message["temperature"] = temp
		message["humidity"] = hum
		message["batterry_pct"] = batt_pct
		message["batterry_mv"] = batt_mv
		os.system('/usr/bin/mosquitto_pub -h 192.168.0.33 -t {} -m \'{}\''.format(topic,json.dumps(message)))
		print('mqtt msg published to: ', topic, ' message: ', message)

	# Blocking call (the given handler will be called each time a new LE
	# advertisement packet is detected)
	parse_le_advertising_events(sock,
					handler=le_advertise_packet_handler,
					debug=False,
					mac_addr=mac_list)
except KeyboardInterrupt:
	disable_le_scan(sock)
```

closed 2020-11-12 by atc1441

JsBergbau, 2020-11-29:
Warning: `temp = int(data_str[22:26], 16) / 10` works only with positive temperatures. With negative temperatures it will report > 6000 °C.
Change it to `temperature = int.from_bytes(bytearray.fromhex(data_str[22:26]),byteorder='big',signed=True) / 10.` and it will also work with negative temperatures.

gytisgreitai, 2020-12-05:
for me this python script just prints value for one time, and then does nothing, what am I missing?

JsBergbau, 2020-12-05:
If you want to read the data via Python, you can use MiTemperature2 / LYWSD03MMC.py from https://github.com/JsBergbau/MiTemperature2#atc-mode-usage

nadnerb33, 2020-12-07:
> for me this python script just prints value for one time, and then does nothing, what am I missing?

@gytisgreitai Does the script end? It should run indefinitely, constantly listening until you stop it.

You should run it with screen so it doesn't stop when you close your SSH session.
I have it running as a service and haven't noticed any issues - but perhaps it stops and restarts without me noticing, I'll have to check.

But maybe just try @JsBergbau's version, it's much more sophisticated and also has a watchdog to re-enter scanning mode (perhaps that explains your issue with my code above)

deepcoder, 2020-12-07:
I wrote this c program to read my BLE sensors, it publishes to MQTT the advertised data. Like the others cited, you need to run it in tmux or screen to keep it alive. I went down this rabbit hole after struggling to get the python ble stuff to install and run in various configs. This just requires bluetooth headers and Paho MQTT libraries. I use it on a RPI centrally located in the house to get good reception from 20 or so sensors.
https://github.com/deepcoder/bluetooth-temperature-sensors

JsBergbau, 2020-12-07:
You can also use nohup <command> to keep it running without using tmux or screen.

gytisgreitai, 2020-12-15:
>Does the script end? It should run indefinitely, constantly listening until you stop it. 

no, it just prints the first value, and then runs, but nothing gets printed out. Anyhow I played around with `py-bluetooth-utils` and started with their example, seems to be working (though yours looks similar, so did not dig what is the difference)

As for JsBergbau solution that simply did not seem to be able to connect
```
pi@raspberrypi:~/MiTemperature2 $ sudo ./LYWSD03MMC.py -d A4:C1:38:F1:1B:B3 --atc
Trying to connect to A4:C1:38:F1:1B:B3
Connection lost
Waiting...
Trying to connect to A4:C1:38:F1:1B:B3
Connection lost
Waiting...
```

JsBergbau, 2020-12-15:
For atc you shouldn't specify a device address. Just use `sudo ./LYWSD03MMC.py --atc --watchdog-timer 5` and it should work fine.