atc1441/ATC_MiThermometer #62: Advertise temperature in F but ESPHome thinks it's C

joeandaverde, 2020-11-01:
Changing the display output to F  seems to advertise in F but ESP home thinks that the value coming from the device is in C and converts. The temperature shown in Home Assistant is 150F.

atc1441, 2020-11-01:
Yes. Changeing it to F does also change the advertising. 
It could be changed so it will only change the display to F

joeandaverde, 2020-11-01:
Yeah, It seems like it should always send in C regardless of what it is displaying.

joeandaverde, 2020-11-01:
atleast for esphome/homeassistant

woody2010, 2020-11-02:
I came here to post the same issue, but I'm fairly new so perhaps its me not knowing I need to do something else.  Do I need to have HA convert the value if I want to have Fahrenheit used on the sensor?  My HA shows 154 degrees when I set the firmware to F.

atc1441, 2020-11-02:
Just made an update so that only the LCD will be F and not all temp values

woody2010, 2020-11-02:
Awesome! Thank you

DiscoJeff, 2020-11-04:
You can also specify in ESPHome that the readings are in F

  - platform: xiaomi_lywsd03mmc
    mac_address: xxxxxxxxxxxxx
    bindkey: "xxxxxxxxxx"
    temperature:
      name: “Guest Bath Temperature"
      unit_of_measurement: "°F"
    humidity:
      name: "Guest Bath Humidity"
    battery_level:
      name: "Guest Bath Sensor Battery Level"

atc1441, 2020-11-06:
Oh thank you for that info, will lea e it for now as it it but good to know

closed 2020-11-12 by atc1441

BuxtonCalvin, 2021-05-10:
I hate to bring up this old topic but I'm hoping you can provide an easy fix for me.  My HA system does not allow underlying values to be converted --such as Celsius to Fahrenheit-- at the app level, only at a given created device level--devices sit on top of the app level data.  I use time series databases and other peripheral data repositories to store values that are pointed at the HA system, so I pull values from the app level.  

Would it be possible to add a UI flag that would allow one to specify whether they wanted the sent temp value to equal the UI value-- as currently, the sent value cannot be changed from Celsius.

Thx for the look.