atc1441/ATC_MiThermometer #63: sleep time for sensor?

michapr, 2020-11-02:
Hi,

what do you think, does it make sense to extend the interval for reading temperature/humidity?

Loop delay time is 7 seconds as I have seen (tested) - readsensor() with wakeup / sleep sensor will be called in every loop.

Will it save battery if check the sensor values only once per minute as sample?

atc1441, 2020-11-02:
That would definitely save a lot of battery power 👍 

The reading of the sensor is the longest time waiting as the i2c uses clockstretching which takes around 10ms where the mcu just waits

michapr, 2020-11-10:
PR:
https://github.com/atc1441/ATC_MiThermometer/pull/73

As I have seen the usage of the sensor will take much battery if wakeup if frequently.
Because there are no needs to check the temperature every 6-7 seconds  I have inserted a loop counter for sensor usage.

Advertising package (one unit 20ms, take the Voltage/10 = mA):
[image]

Sensor usage(one unit 20ms, take the Voltage/10 = mA):
[image]

adv_count  will count the number of the loops in the main procedure until meas_count  was reached.

So the delay could be set to a measurement once per minute as sample instead of every loop, it will save much battery.

second:
Additional was fixed the overflow inside the advertising time condition for now with a counter inside ;)

atc1441, 2020-11-10:
This is great :) 

Will merge it

closed 2020-11-12 by atc1441