atc1441/ATC_MiThermometer #64: Too many advertisements for Custom Type?

animal6496, 2020-11-04:
I've 10 ATC customized thermometers and they report to an MQTT broker via a ESP32 gateway running Tasmota (8.5.1). I've flashed the latest ATC_MiThermometer version and all works well using Advertising Type Custom.  However, I can't get the Advertising Interval to 5 minutes. All thermometers keep Advertising on 2 to 6 second intervals (seems a bit random), and it keeps the gateway quite busy. Also this is likely to drain the batteries on the thermometers. 
I can set via Telink Flasher the "Smiley" and Battery on Display option so setting options on the Thermometer works OK. If I use Advertising Type Mi Tasmota only reports in the Teleperiod the Battery percentage. Not sure how to check the advertising frequency. 
Is the behavior for the Custom advertising type correct ? How can I reduce the frequency of Custom advertisements to once every 5 minutes? Any help welcome.

atc1441, 2020-11-04:
Hi right now the lac interval time is about 2 minutes because of an overflow of the timer, this will be fixed in the future

animal6496, 2020-11-04:
Hi not sure I understand this. So I should be able to set this with a FE0C command which should give 10*12 is 120 seconds. Unfortunately that didn't help it remains 2..6 seconds. Also a more conservative (default setting) results in very frequent updates. So may be I misunderstand the advertisement frequency ...

animal6496, 2020-11-04:
BTW I'm happy to test an update of the firmware if that is required ....

atc1441, 2020-11-04:
So if the counter increases for every of these 2-6seconds ones than it could be because of a temp or humidity change bigger then the setting is.

If the counter does not increase then it is the BLE hardware advertising which will allways be at around 2 seconds as the device needs to make itself visible

animal6496, 2020-11-04:
The reported values are identical so no changes in humidity and temperature. So it is really the BLE hardware presence advertisements which include the hum/temp/battery status values. So I did misunderstand the BLE protocol and I've to filter at the Tasmota Gateway.

atc1441, 2020-11-04:
Ok good to know.

You can use byte 17 as a new message notifier it will increase for every new advertisement

animal6496, 2020-11-04:
OK many thanks. You do an excellent job with great support!

closed 2020-11-12 by atc1441