atc1441/ATC_MiThermometer #66: battery level

michapr, 2020-11-04:
Battery level show 100% for voltage <2200 as I have seen while testing.

I have added the battery symbol for low voltage (<15%) by default, if the show_batt_enabled option (blinking display) is disabled.
In this case all the time will be shown temp/hum value - and batt symbol only if low voltage, similar to the original.

Both things are done in pull request
https://github.com/atc1441/ATC_MiThermometer/pull/65

atc1441, 2020-11-06:
I want to see the battery in percent/blinking so i will not merge this request, if blinking is enabled then the other function is nice to have, it was in the firmware at the beginning but i removed it later could be added again of course. 

When persitence storage is finaly added this can be added as a real setting

michapr, 2020-11-06:
You can use the blinking by default. If you will disable it (by Web UI or compiling with other flag) then the standard function will be enabled.

Second is an issue with low voltage <2200 - here will be shown 100% ;)

michapr, 2020-11-10:
I have checked - "blinking" or any LCD update seems not to be critical for battery usage ;)

atc1441, 2020-11-10:
Thank you again for checking on the blinking powerusage, i was quite shure it does not increase power draw but never checked it

tablatronix, 2021-01-24:
Cycling bat/hum only at threshold would be a nice addition, really do not need to see bat% at 99%, I am running 4 of these, will keep and eye out for battery usage also

atc1441, 2021-01-25:
If the cycling is disabled the battery icon will come up all the time under a fixed threshold in the current firmware, this could be changed to then also display the battery status in %