atc1441/ATC_MiThermometer #67: Longer BLE range?

gabest11, 2020-11-04:
I found in the source that rf_set_power_level_index sets a fixed hard-coded value. According to [this](https://datasheet.lcsc.com/szlcsc/2005281132_LSD-LSD4BT-K53ASTD001_C521558.pdf) document, it corresponds to about 7.425 mA. It would be totally fine for me to increase this to the max (20 mA) and reduce the interval between advertisments to 5 minutes or even longer, as it would be logged by home-assistant anyway. The question is, how much range could I gain by doing this? Is it worth it?

michapr, 2020-11-04:
Don't forget that the device will send every x seconds anyway an advertising package so that it could be visible (used for scanning etc.) 
The provided "advertising time" here is only for advertising changed values in package.

So I think your battery could be empty fast...

BTW: what range do you need there?

atc1441, 2020-11-04:
An option would be to only set the TX power high every x minutes

Then it would not increase the drain a lot

gabest11, 2020-11-04:
I am trying to cover more sensors with a single esp32 in the house. The reception is lost over 10 meters or just in the next room. Uploaded a firmware into one with max strength and currently testing it.

gabest11, 2020-11-05:
It is either placebo or not working. I think it might reach a few meters further in the next room. But only detectable with my notebook, the esp thingy is too weak.

michapr, 2020-11-06:
Try to use ESP32 with OpenMQTTGateway as sample - I have no problems at same floor, also from basement I receive data.
(using ESP32 dev board)

JsBergbau, 2020-11-11:
You could also try a Raspberry PI Zero W. This model has a very far bluetooth range.

marc-gist, 2020-11-15:
> You could also try a Raspberry PI Zero W. This model has a very far bluetooth range.

@JsBergbau have you done this? could you point me to a project/guide showing how to setup a PiZeroW with wifi and ble gateway to mqtt or HA?

michapr, 2020-11-15:
@marc-gist you can use RPIEasy as sample.
https://github.com/enesbcs/rpieasy

But with ESP32 and OpenMQTTGateway it is working also very well.
https://github.com/1technophile/OpenMQTTGateway

JsBergbau, 2020-11-15:
Hi @marc-gist,

you setup a RPI with RPI-OS, I recommend lite version for PI-Zero W. Just flash it to an SD Card, put the wpa_supplicant.conf into boot partition like described here https://www.raspberrypi.org/documentation/configuration/wireless/headless.md and add a file "ssh" without file extension. After that you boot your RPI-Zero W and can access it via Wifi without monitor and keyboard.

After that you install git by sudo apt install git and then you execute
git clone https://github.com/JsBergbau/MiTemperature2
chmod +x *.sh
chmod +x LYWSD03MMC.py 
and then you execute in the folder with ./LYWSD03MMC.py --atc --watchdogtimer 5 --callback sendToMQTT.sh
You must adapt the sendToMQTT.sh for your needs, but it should be quite easy. Just read the https://github.com/JsBergbau/MiTemperature2/blob/master/README.md 
I hope that everything is well documentated.

With that script you can also enable "roaming" with multiple receivers.

elgatobavaria, 2022-10-25:
> An option would be to only set the TX power high every x minutes
> 
> Then it would not increase the drain a lot

Is it somehow possible with the custom firmware ?