atc1441/ATC_MiThermometer #68: Humidity not accurate?

HappySl0th, 2020-11-06:
I am testing out the accuracy of the sensors and placed them inside my humdity controlled drybox. I realised that the original firmware and custom firmware would report different values for humidity (+- 5%). The temperature reported variance is fine at 0.1C, but i set my drybox at 42% humidity (not sure how accurate this is) but original firmware measures it at 50% humidity and custom firmware measures it at 55%. I know with the custom firmware I am able to add an offset. Is anybody having the same issues?

atc1441, 2020-11-06:
The 5% difference is allways there with the custom firmware.

I am using the value directly from the sensor with the custom firmware.
The stock firmware has this offset you see.

I dont knoe which value the right one is as i have no reference value

TheNitek, 2020-11-06:
From my experience I'd say you mixed them up: Stock firmware is usually 5% above the ATC firmware, not the other way around.

closed 2020-11-06 by HappySl0th

JsBergbau, 2020-11-11:
@atc1441 You should mention that your firmware reports another humidity value in youre Readme.md

I've calibrated all my sensors and will do it now again, after I know that there is a difference between.

More about calibration and how to get a reference very easily https://github.com/JsBergbau/MiTemperature2#calibration