atc1441/ATC_MiThermometer #69: Update error: NotSupportedError: GATT operation failed for unknown reason.

AndyCodeX, 2020-11-06:
Error while flashing the stock firmware to go back. Too slow process and random time get this error. Applied stock firmware used: d4135e135443ba86e403ecb2af2bf0af_upd_miaomiaoce.sensor_ht.t2.zip - d4135e135443ba86e403ecb2af2bf0af_upd_miaomiaoce.sensor_ht.t2.bin for LYWSD03MMC. Unable to flash back to stock :(

atc1441, 2020-11-06:
Try it a few times with reconnection it will get faster

AndyCodeX, 2020-11-06:
I have found a solution: close all other opened tabs in a browser while flashing. Process took about 20s and successfully completed. Please add note about browser tabs to fleshing instruction.

atc1441, 2020-11-06:
That should not be a problem, what is a problem is when you switch to another tab while flashing.

This is a chrome "feature" to save ram, it will only render the viewed tabs

AndyCodeX, 2020-11-06:
> That should not be a problem, what is a problem is when you switch to another tab while flashing.
> 
> This is a chrome "feature" to save ram, it will only render the viewed tabs

I know about it but I didn't switch between tabs during flashing. It looks like Chrome shares processor time between tabs (processes) and it make flesher unresponsive from time to time. 
Just a guess...

closed 2020-11-06 by AndyCodeX