atc1441/ATC_MiThermometer #70: Cannot re-connect after flashing custom firmware

elitehammer, 2020-11-08:
I have successfully flashed custom firmware. Changed advertising type to Mi like
Changed advertising interval to 5 minutes

Then proceeded to flash the second thermometer. After I was done with the second thermometer. I decided to switch off smiley and battery update on display on the first one. I can see ATC device when connecting via Telink flasher but logs s the following:
 ...
Reconnect 1 of 5
NetworkError: connection failed for unknown Reason
Reconnect 2 of 5
Detected custom firmware 
...

I tried to activate thermometer via xiaomi home app but as I suspected activation also keept failing (my guess is due to custom firmware) 

My question is will I be able to change those parameter via ESPhome?

atc1441, 2020-11-08:
Thats spund exactly like everything worked as expected. 

The settings can only be set via the web flasher for now

jonastep, 2020-11-19:
Hi,

As far as I understand that means that the flash was correct. What will be the next steps? Adding the device to Mi Home (I cannot find it in MI HOME) and then try to add it to Home assistant declaring "ble_monitor:" in the configuration.yaml ?

elitehammer, 2020-11-19:
Thanks, @atc1441 for the reply. Any suggestions on how to get it connected back to the web flasher as mentioned in OP I keep getting the "connection failed" message?

atc1441, 2020-11-19:
I would suggest to try another device and reboot both beforehand

benleb, 2020-11-19:
I have exactly the same issue :D At least from Chrome (via other tools, the device is discovered within seconds and connecting is no problem)

Arman-GeDon, 2021-01-03:
> I would suggest to try another device and reboot both beforehand

The same problem here.
I uploaded the firmware a month ago to a fresh device with no problem. Today I wanted to connect again (https://atc1441.github.io/TelinkFlasher.html) but no avail. Tried two android smartphones (one was the same as initial) and a PC with Windows 10, changed battery to a new one. 
The site can't connect and stuck on "Detected custom Firmware"
No problem to connect to MHO-C401 or to a new LYWSD03MMC with original firmware.
Log:
22:46:10: Searching for devices
22:46:27: Connecting to: ATC_169C19
22:46:37: Detected custom Firmware
23:05:51: Not connected
23:05:53: Disconnected.
23:05:53: Searching for devices
23:05:55: Connecting to: ATC_169C19
23:05:57: Disconnected.
23:05:57: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
23:05:57: Reconnect 1 from 5
23:06:04: NetworkError: Connection failed for unknown reason.
23:06:04: Reconnect 2 from 5
23:06:12: NetworkError: Connection failed for unknown reason.
23:06:12: Reconnect 3 from 5
23:06:18: Detected custom Firmware

atc1441, 2021-01-03:
Hey. For me it looks exactly how it is supose to be. There should not happen anything after Detected custom firmware, you can now do some settings :) 

What are you having problems with ?

Arman-GeDon, 2021-01-03:
Thanks.
I think there should be "Connected" at the end - that's what I have on  MHO-C401 or on a new LYWSD03MMC with original firmware. Logs then looks like these:
02:02:49: Searching for devices
02:02:52: Connecting to: LYWSD03MMC
02:02:54: Detected Mi Thermometer
02:02:54: Connected

But on the flashed LYWSD03MMC the log ends on "Detected custom Firmware". Then when click "Do Activation" there is "Not connected"

But I think I found solution: 
You have to click on some options (Smiley, LCD, advertising ) then connect again and device is connected  (in the log) and you do activation.
BTW: What advertising interval do you recommend?

atc1441, 2021-01-03:
You can of course not do an activation of the custom rom as it does not habe any encryption like the stock :) 

The best interval is 1 minute as it does not drain the battery more and the you have the fastet update rate

Arman-GeDon, 2021-01-03:
Ok, maybe my stupid mistake because it's late :)
Thanks for help and thanks for your work!

gitkobold, 2021-01-15:
Hi @atc1441 ,

Is it possible to flash the fw to the device with nrf Android app? (Is it used the a standard method?)
Thanks

atc1441, 2021-01-15:
Hey. No the protocol is an Telink propriatary one, also the activation needs to be done before flashing ota

alexig89, 2023-04-26:
> You can of course not do an activation of the custom rom as it does not habe any encryption like the stock :)
> 
> The best interval is 1 minute as it does not drain the battery more and the you have the fastet update rate

Hello and thank you for an answer - I tried to find it since yesterday :)
But I have another question - how to add the sensor to Home Assistant if I haven't Mi Bind Key and my Mi thermometer has custom firmware?