atc1441/ATC_MiThermometer #71: advertising interval

michapr, 2020-11-08:
Checking the battery usage I have seen that the advertising interval is about 1,9s instead of 3s (if understand the setting in app_config.h right).
`#define ADVERTISING_INTERVAL 3000`

If change it to 6000 the advertising time is about 3,8 seconds - this will save battery, right?
setting 3000 (one unit is 1sec):
[image]
setting 6000 (one unit is 1sec):
[image]

Do not look at the noise, I did not make any filter... ;)

atc1441, 2020-11-08:
Hey. 

Yes that will increase battery life times 2 

The advertising time is calculated by 0.65 if i remember correctly. Thats why its such a strange number

michapr, 2020-11-08:
I have changed additional the sensor wakeup time - it will give more battery time.
What do you think, make it configurable for user (web UI) - or is a setting in code fine?
like 
`RAM uint8_t measure_interval = 6;//multiply by 10 for value in sec.`

Will make a PR in next days with some pictures like above ;)

atc1441, 2020-11-08:
I would only make that setting in code as you can lock yourself out of the thermometer with a wrong setting

michapr, 2020-11-08:
> Otherwise it is a good feature. But bear in mind the longer the interval the harder it is to connect to the device

Right. The original device have nearly the same advertising time. I think take 3 seconds would be ok.

For temperature measure interval I think to take 1 minute by default - just now it is about 7,5 seconds.
It will save much battery because the measure time is relatively long, as you already said.

michapr, 2020-11-10:
I think it maybe a feature for special optimization, but not for everybody now. 
Anyway I think it could be changed to 3 seconds (x2) to save battery on one side and reach it stable... Will be within standard.

**The advertising interval should be in the interval 20 ms to 10.24 s** (Bluetooth Core Specification v 4.2, Vol 2, Part E, chapter 7.8.5).

twocolors, 2020-11-26:
@michapr hi, i have lost battery every day ~ 1-2% , rssi > 78

If this is too much just ignore but could you compile for me as fallow: Smile: Comfort indicator, Custom, Battery: Enabled, Advertise: 180 seconds (3m) and up advertising interval ble to ~ 7 sec

michapr, 2020-11-26:
@twocolors 
hi, I will make it for you, I also try to get less battery usage...
The advertise 180 seconds will not help, this means only that any detected change will be reported after 180 seconds.
I will make "next measurement" (enable sensor) every 180 seconds then for you - ok?
see https://github.com/atc1441/ATC_MiThermometer/issues/63

twocolors, 2020-11-26:
@michapr thx, for help. 180 seconds for "next measurement" - ok.
It all works with the help of magic, i have three sensors
1. rssi > 70 < 75 , 7 days - 100% battery
2. rssi > 75 < 80 , battery every day ~ -1-2%
3. rssi > 75 < 80 , battery every day ~ -1-2%

i think will help if "advertising interval" up to 7-9 sec, to not poll the sensor so often

michapr, 2020-11-26:
@twocolors 
I made two images for you - one with about 6 seconds BLE advertising interval, second with about 8seconds
(in case of problems with your requesting software. - what do you use?).
Additional the battery measurement is more limited - will be one only every 50th loop, while temperature is measured.
So every about 150-160 seconds * 50 = 7500seconds = every 2 hours about. This all will save battery...

Your files:
https://www.smarthome-work.com/download/ATC_Thermometer__batt_6sec-adv_3min-meas_opt-batt-meas.bin
https://www.smarthome-work.com/download/ATC_Thermometer__batt_8sec-adv_3min-meas_opt-batt-meas.bin

Tell me please about the battery usage after some time ;)

BTW: all batteries are new - and from same date / manufacturer?

twocolors, 2020-11-26:
i use https://github.com/espruino/EspruinoHub , yes all battery replace ikea CR2032 i check and have 3.3V

twocolors, 2020-11-26:
wow in you fw all battery > 85% && < 90% level, 8sec is really 7.3
thx, i will watch, maybe the problem is in the quality of the battery

michapr, 2020-11-26:
After flashing the battery is going a bit down. Flashing take some energy.

michapr, 2020-12-05:
@twocolors 
Any changes in battery usage visible?

twocolors, 2020-12-13:
@michapr oh ... show ...

[image]

michapr, 2020-12-14:
seems better as before (was 1-2% per day)?

Will send you another image in a few days, with additional energy optimizations ;)
(it is just in test at my sensors)

twocolors, 2020-12-14:
yes, 1-2% per day

i use ATC_Thermometer__batt_8sec-adv_3min-meas_opt-batt-meas.bin

AndrewLinden, 2020-12-14:
> I would only make that setting in code as you can lock yourself out of the thermometer with a wrong setting

Would it be possible to vary the advertising interval like so:
Upon cold boot for the first couple of minutes a small advertising interval is used to make it possible to connect to the module.
After that a user configurable long advertising interval (like 300s or 600s) would be used for maximum battery saving.
?

hancockks, 2020-12-16:
@AndrewLinden that's a really good idea!  Should be easy to do with just a counter.

twocolors, 2021-01-22:
@michapr hi, is there something to test? or you can go to build from https://github.com/atc1441/ATC_MiThermometer/releases/tag/30

michapr, 2021-01-23:
Hi,

you can use this forked repo - here is all optimized inside ;)
https://github.com/pvvx/ATC_MiThermometer/

Michael