atc1441/ATC_MiThermometer #74: Defauld temp

cobra85ec, 2020-11-10:
hey is there a way to make the default temp show as F after removing the battery  ... thx   i also made a donition thank 4 u work

atc1441, 2020-11-10:
Hey  thank you.

Please take a look at this issue maybe it will help
https://github.com/atc1441/ATC_MiThermometer/issues/37

cobra85ec, 2020-11-10:
that didnt help  link no longer work

cobra85ec, 2020-11-10:
> Hey thank you.
> 
> Please take a look at this issue maybe it will help
> #37

can u plz help me  thats all i would need

atc1441, 2020-11-11:
The fastest way for you to archive it is to compile your own version with F as default. 

I can do it in the next days and upload it here but it will take some time

atc1441, 2020-11-11:
Here it is for you:

please unpack before flashing

[ATC_Thermometer_F.zip](https://github.com/atc1441/ATC_MiThermometer/files/5522033/ATC_Thermometer_F.zip)

closed 2020-11-12 by atc1441

cobra85ec, 2020-11-13:
> Here it is for you:
> 
> please unpack before flashing
> 
> It is not tested and just compiled so do it at your own risk
> 
> [ATC_Thermometer_F.zip](https://github.com/atc1441/ATC_MiThermometer/files/5522033/ATC_Thermometer_F.zip)

thank u very much... im trying to learn to build custom fw  but its not easy